MiCode/Xiaomi_Kernel_OpenSource #993: whyred touchscreen firmware update for tulip

thesprintster, 2018-12-22:
whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
Please update whyred kernel with tulip changes.

rama982, 2018-12-22:
Yonwtf

yuvrajx, 2018-12-22:
> whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
> Please update whyred kernel with tulip changes.

FYI,
For every touchscreen delays and touchscreen data differs.
I can help you but that'll be pointless

Haseo97, 2018-12-23:
pLeAsE uPdAtE WhYrEd kErNeL wItH tULiP ChAnGeS

Xiaomi: Yo wtf are u asking dude

yuvrajx, 2019-01-03:
> whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
> Please update whyred kernel with tulip changes.

Is both device have same touchscreen model?

rama982, 2019-01-03:
@yuvrajx actually no

yuvrajx, 2019-01-03:
> whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
> Please update whyred kernel with tulip changes.

If I have time I'll try to adapt touchscreen driver for your device you can ping me on [telegram](https://t.me/yuvrajx), the problem is that there's many drivers that one device xiaomi can ship.
And unfortunately I can't do all of them due to lack of time I'll try if I have sometime to figure this out and as you have mentioned above that it boots without touchscreen than it can boot after adapting the needed touchscreen drivers.

rama982, 2019-01-03:
@yuvrajx actually, tulip novatek driver is 'accidentaly' pushed on jasmine-p-oss already, with some other stuff too.
Even i already rebased it and make boot of it on my phone.
What left is tulip focaltech driver. On jasmine-p-oss missing firmware .i file.

yuvrajx, 2019-01-03:
> @yuvrajx actually, tulip novatek ts driver is 'accidentaly' pushed on jasmine-p-oss already, with some other stuff too.
> Even i already rebased it and make boot of it on my phone.
> What left is tulip focaltech ts driver. On jasmine-p-oss missing firmware .i file.

I can help you
Let's see you on telegram I can fix it :)

yuvrajx, 2019-01-04:
> whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
> Please update whyred kernel with tulip changes.

Can you try flash this kernel?
Make sure you flash it on custom roms
https://drive.google.com/a/my.cuesta.edu/uc?id=1h_i0lAryMNmCIoNDgSQmTspkpIDLm1fI&export=download

yuvrajx, 2019-01-04:
> whyred stock 10.0.4.0 kernel boots in tulip but touch is not working.
> Please update whyred kernel with tulip changes.

Can you try flash this [kernel](https://drive.google.com/a/my.cuesta.edu/uc?id=1h_i0lAryMNmCIoNDgSQmTspkpIDLm1fI&export=download)
Make sure you flash it on custom roms.

thesprintster, 2019-01-09:
> pLeAsE uPdAtE WhYrEd kErNeL wItH tULiP ChAnGeS
> 
> Xiaomi: Excuse me waduhek

@Haseo97 Take the L!
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/efbbb9cce47475c49f28d227b2a52eca6f0d02a7

closed 2019-01-09 by thesprintster

Haseo97, 2019-01-09:
Yes i’m taking the L now