MiCode/Xiaomi_Kernel_OpenSource #996: Please release Android P kernel source for Mi 8 SE (sirius)

longnt2007, 2018-12-28:
As Mi8 SE china beta rom have been upgraded to Android 9.0 nearly two months, please release Android P kernel source for Mi 8 SE (sirirus) too. Also please check the DTB building config as default config doesn't work correctly, enable CONFIG_BUILD_ARM64_DT_OVERLAY will show error when building source, disable CONFIG_BUILD_ARM64_DT_OVERLAY (as default config) is ok but dtbs cooked is not compatible with stock dtbo.img partition.

Thank you.

Titokhan, 2019-03-08:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/grus-p-oss

closed 2019-03-09 by longnt2007

rama982, 2019-03-09:
Mi8 SE Pie kernel source already released
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/b18c42adc8a20c2775cc55f079b2bea047674a4d

dkmgfk, 2019-03-10:
longnt2007

Hello! Do you have any plan for porting AOSP based ROM to this device?

Regards