MiCode/Xiaomi_Kernel_OpenSource #1020: Release source of Redmi 4 (nikel)

Initiatoro, 2019-01-16:
Oh C'mon
You released sources of note 2 and 3 (mtk)

please release of note 4 (mtk) also 

hoping for reply (and kernel source also :3)

dragonpt, 2019-01-17:
It's on the " warehouse" 

It's expected soon

moham96, 2019-03-26:
@dragonpt Any update ? 
I really need the source, even if it's without fingerprint reader drivers and camera sensors for now, just need the kernel and the device tree please.
Regards

Yardanico, 2019-10-15:
@dragonpt any update on "soon"? :)
It's really a strange situation, the device was released over 3 years ago, it's almost nonexistent , why does Xiaomi not release the sources - just release the sources and forget the device :D

barfin, 2020-01-21:
> @dragonpt any update on "soon"? :)
> It's really a strange situation, the device was released over 3 years ago, it's almost nonexistent , why does Xiaomi not release the sources - just release the sources and forget the device :D

one year later and still we're not close to "soon"

Yardanico, 2020-01-22:
@barfin yeah, that's sad :(

barfin, 2020-01-25:
> @barfin yeah, that's sad :(

i hate xiaomi and mediatek for this
they're violating gpl license

lex3a, 2020-03-04:
I won't never buy any Mediatek device again. 
Release the source code!

nkhdiscovery, 2020-09-18:
> It's on the " warehouse"
> 
> It's expected soon

@dragonpt Any new plans? Years have been passed!

Waxaranai, 2021-02-07:
No update? *sigh*

Ruben1863, 2022-01-15:
Hey, I recently booted the custom kernel for xiaomi nikel. I managed to decode correctly one of three lcms from stock kernel. The other two lcms also possible to decode "easily". For now I just have display working (and just nt35532_fhd_boe_vdo_lcm). If any developer wants to help, contact me. 
My kernel source: https://github.com/Ruben1863/android_kernel_xiaomi_nikel
Note: For blind people, this is my custom kernel, not Xiaomi official ones.