MiCode/Xiaomi_Kernel_OpenSource #1024: Chiron (sagit?) smmu warning

wight554, 2019-01-29:
All chiron (sagit?) builds after 8.10.20 have this warning due to tz update
Any chance to get this fixed?

`[    1.429682] ------------[ cut here ]------------
[    1.429688] WARNING: at /home/work/chiron-o-dev-build/kernel/msm-4.4/drivers/iommu/arm-smmu.c:2499
[    1.429695] Modules linked in:
[    1.429704] 
[    1.429716] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.4.78-perf-g5f78432 #1
[    1.429722] Hardware name: Qualcomm Technologies, Inc. MSM 8998 v2.1 MTP (DT)
[    1.429729] task: ffffffc176688000 ti: ffffffc176690000 task.ti: ffffffc176690000
[    1.429747] PC is at arm_smmu_assign_table+0xb0/0x134
[    1.429753] LR is at arm_smmu_assign_table+0xa8/0x134
[    1.429758] pc : [<ffffff8008b65a60>] lr : [<ffffff8008b65a58>] pstate: 60000145
[    1.429763] sp : ffffffc1766934e0
[    1.429767] x29: ffffffc1766934e0 x28: 0000000000000000 
[    1.429776] x27: 0000000000000020 x26: ffffffc17499a618 
[    1.429786] x25: ffffffc171510ec8 x24: ffffff8009c06000 
[    1.429795] x23: ffffff8009c06000 x22: ffffffc171510e00 
[    1.429804] x21: ffffffc171510f00 x20: 00000000ffffffea 
[    1.429814] x19: ffffffc1714fc600 x18: ffffffc176693468 
[    1.429823] x17: 00000000b51ace00 x16: 000000002d30ed20 
[    1.429832] x15: 00000000000f4240 x14: 746365746f727020 
[    1.429840] x13: 79726f6d656d206e x12: 6769737361206f74 
[    1.429850] x11: 2064656c69614620 x10: 3a656c6261745f6e 
[    1.429859] x9 : 67697373615f7079 x8 : ffffff80082e66f8 
[    1.429868] x7 : ffffff8009c538f0 x6 : ffffff800a50f69f 
[    1.429877] x5 : ffffff800a501000 x4 : 0000000000000000 
[    1.429886] x3 : 0000000000025b48 x2 : 0000000000000000 
[    1.429895] x1 : ffffff8009a8ca40 x0 : 00000000ffffffea 
[    1.429906] \x0aPC: 0xffffff8008b65960:
[    1.429911] 5960  321e0021 36600163 f9401262 8b020000 97e6dea0 34000100 f9400260 900063e1
[    1.429940] 5980  911f6821 97ea3ab1 128002a0 14000006 b9000041 52800000 34000074 aa1303e0
[    1.429965] 59a0  97ffffb5 a94153f3 a8c27bfd d65f03c0 a9ba7bfd 52800061 910003fd a90153f3
[    1.429991] 59c0  a9025bf5 a90363f7 aa0003f6 b90053a1 b940f802 52800014 b90057a2 528000c2
[    1.430016] 59e0  b9005ba2 52800082 b9004fa1 b9005fa2 97fffe6b 53001c00 340006a0 f94082d3
[    1.430042] 5a00  910402d5 b0008517 b0008518 d1004273 91004260 eb0002bf 540002e0 f9400260
[    1.430067] 5a20  b63000a0 f9451b01 92409400 8b010000 14000003 f94522e1 cb010000 d2820001
[    1.430092] 5a40  910133a2 52800023 910143a4 910163a5 52800046 97e741fb 2a0003f4 34000060
[    1.430118] 5a60  d4210000 14000005 f9400a73 d1004273 17ffffe8 52800014 f94082c0 f85f0413
[    1.430144] 5a80  d1004273 91004001 eb0102bf 54000200 f9400802 f9400c01 f9000441 f9000022
[    1.430169] 5aa0  d2802001 f2fbd5a1 f9000801 d2804001 f2fbd5a1 f9000c01 97d904b2 f9400a61
[    1.430194] 5ac0  aa1303e0 d1004033 17ffffef 2a1403e0 a94153f3 a9425bf5 a94363f7 a8c67bfd
[    1.430218] 5ae0  d65f03c0 a9bb7bfd 528000e2 52800061 910003fd a90153f3 a9025bf5 f9001bf7
[    1.430243] 5b00  aa0003f4 b90047a2 b940f802 b90043a1 b9004ba2 b9004fa1 97fffe21 53001c00
[    1.430268] 5b20  340006c0 f9408a93 91044295 b0008516 b0008517 d1004273 91004260 eb0002bf
[    1.430293] 5b40  54000320 f9400260 b63000a0 f9451ae1 92409400 8b010000 14000003 f94522c1
[    1.430318] \x0aLR: 0xffffff8008b65958:
[    1.430323] 5958  b9400041 b9403663 321e0021 36600163 f9401262 8b020000 97e6dea0 34000100
[    1.430348] 5978  f9400260 900063e1 911f6821 97ea3ab1 128002a0 14000006 b9000041 52800000
[    1.430372] 5998  34000074 aa1303e0 97ffffb5 a94153f3 a8c27bfd d65f03c0 a9ba7bfd 52800061
[    1.430398] 59b8  910003fd a90153f3 a9025bf5 a90363f7 aa0003f6 b90053a1 b940f802 52800014
[    1.430423] 59d8  b90057a2 528000c2 b9005ba2 52800082 b9004fa1 b9005fa2 97fffe6b 53001c00
[    1.430448] 59f8  340006a0 f94082d3 910402d5 b0008517 b0008518 d1004273 91004260 eb0002bf
[    1.430472] 5a18  540002e0 f9400260 b63000a0 f9451b01 92409400 8b010000 14000003 f94522e1
[    1.430497] 5a38  cb010000 d2820001 910133a2 52800023 910143a4 910163a5 52800046 97e741fb
[    1.430523] 5a58  2a0003f4 34000060 d4210000 14000005 f9400a73 d1004273 17ffffe8 52800014
[    1.430548] 5a78  f94082c0 f85f0413 d1004273 91004001 eb0102bf 54000200 f9400802 f9400c01
[    1.430574] 5a98  f9000441 f9000022 d2802001 f2fbd5a1 f9000801 d2804001 f2fbd5a1 f9000c01
[    1.430599] 5ab8  97d904b2 f9400a61 aa1303e0 d1004033 17ffffef 2a1403e0 a94153f3 a9425bf5
[    1.430624] 5ad8  a94363f7 a8c67bfd d65f03c0 a9bb7bfd 528000e2 52800061 910003fd a90153f3
[    1.430650] 5af8  a9025bf5 f9001bf7 aa0003f4 b90047a2 b940f802 b90043a1 b9004ba2 b9004fa1
[    1.430674] 5b18  97fffe21 53001c00 340006c0 f9408a93 91044295 b0008516 b0008517 d1004273
[    1.430699] 5b38  91004260 eb0002bf 54000320 f9400260 b63000a0 f9451ae1 92409400 8b010000
[    1.430726] \x0aSP: 0xffffffc1766933e0:
[    1.430730] 33e0  0a50f69f ffffff80 09c538f0 ffffff80 082e66f8 ffffff80 615f7079 67697373
[    1.430756] 3400  61745f6e 3a656c62 69614620 2064656c 61206f74 67697373 656d206e 79726f6d
[    1.430781] 3420  6f727020 74636574 000f4240 00000000 2d30ed20 00000000 b51ace00 00000000
[    1.430807] 3440  76693468 ffffffc1 714fc600 ffffffc1 ffffffea 00000000 71510f00 ffffffc1
[    1.430832] 3460  71510e00 ffffffc1 09c06000 ffffff80 09c06000 ffffff80 71510ec8 ffffffc1
[    1.430857] 3480  7499a618 ffffffc1 00000020 00000000 00000000 00000000 766934e0 ffffffc1
[... 35 lines cut ...]
[    1.431410] [<ffffff80085f800c>] driver_probe_device+0x1a4/0x400
[    1.431417] [<ffffff80085f83b4>] __device_attach_driver+0x88/0x98
[    1.431425] [<ffffff80085f6224>] bus_for_each_drv+0x70/0x8c
[    1.431431] [<ffffff80085f7d68>] __device_attach+0x90/0x11c
[    1.431437] [<ffffff80085f8520>] device_initial_probe+0x10/0x18
[    1.431444] [<ffffff80085f725c>] bus_probe_device+0x2c/0x8c
[    1.431452] [<ffffff80085f5364>] device_add+0x2fc/0x53c
[    1.431465] [<ffffff80089d68d8>] of_device_add+0x38/0x40
[    1.431471] [<ffffff80089d70ac>] of_platform_device_create_pdata+0x98/0xd0
[    1.431477] [<ffffff80089d73a4>] of_platform_bus_create+0x2a4/0x364
[    1.431483] [<ffffff80089d75e0>] of_platform_populate+0x70/0xac
[    1.431490] [<ffffff8008885ed8>] msm_vidc_probe+0x9fc/0xb6c
[    1.431496] [<ffffff80085f9ef8>] platform_drv_probe+0x40/0x8c
[    1.431502] [<ffffff80085f800c>] driver_probe_device+0x1a4/0x400
[    1.431509] [<ffffff80085f82cc>] __driver_attach+0x64/0x90
[    1.431515] [<ffffff80085f6e38>] bus_for_each_dev+0x74/0x90
[    1.431522] [<ffffff80085f79d4>] driver_attach+0x20/0x28
[    1.431529] [<ffffff80085f7550>] bus_add_driver+0x13c/0x24c
[    1.431536] [<ffffff80085f8ea4>] driver_register+0x94/0xe0
[    1.431543] [<ffffff80085f9e2c>] __platform_driver_register+0x48/0x50
[    1.431552] [<ffffff8009a459d4>] msm_vidc_init+0x134/0x1c0
[    1.431562] [<ffffff8008082bb8>] do_one_initcall+0x190/0x1ac
[    1.431577] [<ffffff8009a00db4>] kernel_init_freeable+0x1a4/0x244
[    1.431591] [<ffffff8008eb83e0>] kernel_init+0x10/0xf8
[    1.431599] [<ffffff80080826f0>] ret_from_fork+0x10/0x20`

wloot, 2019-01-29:
Yes, sure. It also happens on sagit.

ivan19871002, 2019-01-30:
Sorry ， I feed it back inside

sekaiacg, 2019-02-04:
Yes, there may be more
[image]

wight554, 2019-02-04:
That's not stock kernel and more like ur own issue