MiCode/Xiaomi_Kernel_OpenSource #1026: Missing braces error and undefined references Redmi 6/6A

Dhoine, 2019-02-07:
Redmi 6/6A (cactus-o-oss)
Undefined references to 
`extern struct spm_lp_scen __spm_dpidle;
extern struct spm_lp_scen __spm_sodi3;
extern struct spm_lp_scen __spm_sodi;`
See temp solution (seems, that it doesn't affect device's work): https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/239f62196a6d5ba58d92df195aca43e2ea884a78
Missing drivers for these modes?
Missing braces - error appears while building with google's 4.9 androideabi toolchain : see https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/c6428ea4b6e10d1eefb8d94b466e5137377d7cb1

SamarV-121, 2019-02-07:
Use Linaro GCC 6.3-2017.05

Dhoine, 2019-02-07:
Ok, maybe this will fix braces. Bud udefined references during linkink... I don't think so.