MiCode/Xiaomi_Kernel_OpenSource #1027: Redmi 6/6A (cactus-o-oss) broken kernel modules

Dhoine, 2019-02-07:
Builded kernel can't load modules from /vendor/lib/modules (broken wifi, gps, fm, etc.). Even with force load (insmod says wrong exec format). And sources for these modules (and fpsgo.ko from lib/modules) are not included into kernel sources (vendor/mediatek/kernel_modules). Please, move sources to kernel...

SamarV-121, 2019-02-07:
Yep ! exactly 
need module source

rama982, 2019-02-07:
or find hacky way to use prebuilt miui modules

Dhoine, 2019-02-07:
I tried enabling modversions — no result.
I tried enabling force modules load — no result. Insmod always gives me wrong exec format error.
With what toolchain was stock kernel built? Maybe it's a key...

fpgadsp, 2019-02-10:
> Builded kernel can't load modules from /vendor/lib/modules (broken wifi, gps, fm, etc.).
> Even with force load (insmod says wrong exec format). And sources for these modules (and fpsgo.ko from [...]
hi Dhoine，Did you compile the kernel with alps?  Could you tell me how to compile？

Dhoine, 2019-02-10:
I didn't compile these sources with alps. It's useless, all references to mtk vendor were removed. I've found references only on mtk's generic sources from alps (I can't share link to alps, sorry, the guy who uploaded it don't want this)
I've seen your issues here. The kernel builds as any generic android kernel. With google toolchain.

Dhoine, 2019-02-21:
Fixed by hacking kernel/module.c to bypass module sympols crc check... Haven't found another way.

minhducsun2002, 2020-10-04:
The Pie version seems to work with community patches, but are there any progress with kernel modules?

Dhoine, 2020-10-04:
As far as I remember, Pie kernel sources don't have module sources. So, the problem remains. I haven't tried it, cause my first attempt with it costed me $10 for unbricking the device. I stopped further attempts.

minhducsun2002, 2020-10-23:
@Dhoine you might want to log into XDA again and check the old thread. I found a solution to this problem.

setyanzxc123, 2023-05-27:
> Fixed by hacking kernel/module.c to bypass module symbols crc check... Haven't found another way.

how ?