MiCode/Xiaomi_Kernel_OpenSource #1030: [daisy-o-oss] broken wifi

eremitein, 2019-02-11:
Error on Xiaomi Mi A2 Lite

kernel compiled complete with commands

$ mkdir out
$ export ARCH=arm64
$ export SUBARCH=arm64
$ export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-android-
$ make O=out daisy-perf_defconfig
$ time make -j$(nproc) O=out 2>&1 | tee kernel.log

then kernel installed on phone, booted and  all works BUT WIFI NOT work!

"pronto" wi-fi module compiled in kernel by default kernel config. and i don't understand why this not work. i need help to solve this problem

eremitein, 2019-02-11:
OK, i found driver via https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ

integrated in sources but now kernel don't complile due error in msm encryption

eremitein, 2019-02-11:
Xiaomi, what the ***!!!

Why didn't you put the source code and instructions with which could build the same kernel as on the stock firmware?..

What you published is not a complete code, you using something else for your development

This is a violation, do you know about it?

rama982, 2019-02-11:
Actually, you can see other xioami custom kernel repo with msm8953 and pick commit for your issue.

eremitein, 2019-02-11:
ok, i see you have included prima)
https://github.com/rama982/android_kernel_xiaomi_msm8953/blob/c47202644dae47eda3181790e87383c300f627e0/drivers/staging/prima/Kconfig

so i will try build your kernel base. thanks

LinuxPanda, 2019-02-11:
You can use this, https://github.com/khusika/android_vendor_qcom_opensource_wlan_prima.

eremitein, 2019-02-11:
@LinuxPanda, that's exactly what i tried^^
and get errors with broken compilation

LinuxPanda, 2019-02-11:
You need to add configs.

* https://github.com/LinuxPanda/android_kernel_rosy/commit/7be0ae1653b69870bce9307f30726743494b4584 [ignore the .gitmodule]

* https://github.com/LinuxPanda/android_kernel_rosy/blob/o-318/arch/arm64/configs/rosy-doge_defconfig#L708

eremitein, 2019-02-11:
of course i did it, without configs kernel not see driver to build^^

--- drivers/staging/Kconfig.orig	2018-12-14 06:33:20.000000000 +0500
+++ drivers/staging/Kconfig	2019-02-11 18:25:49.483081911 +0500
@@ -108,4 +108,6 @@
 
 source "drivers/staging/unisys/Kconfig"
 
+source "drivers/staging/prima/Kconfig"
+
 endif # STAGING
--- drivers/staging/Makefile.orig	2019-02-11 18:41:19.865526332 +0500
+++ drivers/staging/Makefile	2019-02-11 18:41:38.393976627 +0500
@@ -47,4 +47,5 @@
 obj-$(CONFIG_UNISYSSPAR)	+= unisys/
 obj-$(CONFIG_KTRACE)		+= ktrace/
 obj-$(CONFIG_RTMM)			+= rtmm/
+obj-$(CONFIG_PRIMA_WLAN)			+= prima/

eremitein, 2019-02-11:
after all i get error 

<mach/dma.h> not found when drivers/crypto/msm/qce.c build

eremitein, 2019-02-12:
@rama982 hello

why option disabled by default "Enable the Prima WLAN BT-AMP feature" ?

if i enable it kernel don't significally decrease wifi speed with turned bt?
many people have this problem on stock after updating to Pie

TheInfected, 2019-02-13:
> then kernel installed on phone, booted and all works BUT WIFI NOT work!
> "pronto" wi-fi module compiled in kernel by default kernel config. and i don't understand why this not work. i need help to solve this problem
* Attach kernel logs

> after all i got error
> <mach/dma.h> not found when drivers/crypto/msm/qce.c building

* cherry-pick: https://github.com/android-linux-stable/polaris/commit/9abc7e414a0157e11118f9a740e20bef48eb43a9#diff-fd15a6bb0242f341065152ae2dd3cd35

Matheus-Garbelini, 2019-03-31:
Anyone knows where to find code documentation of the prima/pronto wifi drivers? I'm trying to modify it but it seems impossible if you don't work at qualcomm

nysadev, 2019-03-31:
You all should use qcacld-3.0 if you're not

Matheus-Garbelini, 2019-04-01:
@nysadev. Thanks, actually I was compiling and modifying it as they seems the same. But asked about prima because it seems in more active development. In any case do you know some documentation? I look at those folders structures I have no idea what the mean other than HDD is supposed to mean just device drive. They indeed don't make things easy ha? Hehe

nysadev, 2019-04-02:
> @nysadev. Thanks, actually I was successfully compiling and modifying it as they seems the same. But I've asked about prima because it seems in more active development. In any case do you know some documentation? I look at those folders structures I have no idea what the mean other than HDD is supposed to mean just host device driver.
> 
> They indeed don't make things easy ha? Hehe

I don't know about documentation for either. What is it you're trying to find out?

Matheus-Garbelini, 2019-04-02:
@nysadev thanks for your attention.

I'm actually trying to understand the data path for RX so I can sniff raw packets via a custom netlink socket.

So for here's what I understand about the qca-cld3.0:
* There's 2 threads, 1 for mac802.11 management processing and other for data. (cds_mc_thread and cds_ol_rx_thread). 
* Trying to enable promiscuous mode using hdd_conf_hw_filter_mode function (or other low level functions via wma_conf_hw_filter_mode)
* How to enable promiscuous mode for data packets, it seems to be handles by other config than the "gHwFilterMode"
* When changing the mac address via "sme_set_custom_mac_addr" I can receive management packets but the driver seems to throttle packets if I send to much. Changing the mac address of the **sender** resolves. If so, is there an option on the driver ot in the hardware controlling this?

Here's what I've achieved so far:
* Overriding channel and mac address in hotspot mode
* Sniff correctly management frames (even malformed ones)

nysadev, 2019-04-08:
> @nysadev thanks for your attention.
> 
> I'm actually trying to understand the data path for RX so I can sniff raw packets via a custom netlink socket.
> 
> So for here's what I understand about the qca-cld3.0:
> 
> * There're 2 threads, 1 for mac802.11 management processing and other for data. (cds_mc_thread and cds_ol_rx_thread).
> * It seems that promiscuous mode can be partially enabled by  "gHwFilterMode" and using hdd_conf_hw_filter_mode function (or other low level functions via wma_conf_hw_filter_mode)
> * When changing the mac address via "sme_set_custom_mac_addr"
> * I can receive management packets but the driver seems to throttle packets if I send to much. Changing the mac address of the **sender** resolves. If so, is there an option on the driver or in the hardware controlling this?
> * The data thread seems to send the data to hdd_gro_flush. However couldn't understand what format the packet is in this function.
> 
> Here's what I've achieved so far:
> 
> * Overriding channel and mac address in hotspot mode
> * Sniff correctly management frames (even malformed ones)
> 
> @nysadev Sorry for the technical stuff, but I didn't find anyone that could help me with some technical questions about this driver. I've already made successful modifications in the rt2800usb driver, but this one seems way more complex. Hope you have some idea whats going on.
> 
> Thanks.

Sorry, I'm completely useless for this kind of stuff. I'm a kernel noob and all I know is that I should import qcacld-3.0 :P 

Sorry about that.

0xXA, 2019-04-09:
>
> However couldn't understand what format the packet is in this function.
> 

Binary format of the obviously

Matheus-Garbelini, 2019-04-09:
@0xXA haha. I'm talking about the correct struct that the void buffer is referring to. hdd_gro_flush receives a skb data has its data as a *void pointer, which mean it can be anything. I know it's binary, I was trying to figure out what are the fields encoded there so I could get just the raw 802.11 data and not padding or unrelated headers.