MiCode/Xiaomi_Kernel_OpenSource #1031: Redmi 6/6a LK source

p0lishit, 2019-02-15:
Please xiaomi release redmi 6/6a lk source :pray:

luk1337, 2019-02-15:
your wish lol

TheInfected, 2019-02-15:
> Please xiaomi release redmi 6/6a lk source
> or at least apply this changes..
> 
> ```
> --- a/platform/mt6765/platform.c	2019-02-16 04:53:15.751000553 +0700
> +++ b/platform/mt6765/platform.c	2019-02-16 04:52:51.611000552 +0700
> @@ -700,6 +700,28 @@
>  	time_boot_mode = get_timer(0);
>  #endif
>  	boot_mode_select();
> +#ifdef MTK_FASTBOOT_SUPPORT
> +    /* emergency check to prevent brick. in the following case: */
> +    /* a. boot/recovery partition was corrupt/not bootable. */
> +    /* b. rtc recovery flags was set but recovery image is corrupt/not bootable. */
> +    /* c. para partition was set to boot into recovery but recovery image is corrupt/not bootable. */
> +    /* d. odmdtbo was corrupt / odmdtbo somehow cannot be merged to base device tree on custom kernel. */
> +    
> +    /* without this patch the only solution for bricked device is using spflashtool to restore fw which require authorized id */
> +    /* R.I.P "system has been destroyed" */
> +    // translated by google.
> +    
> +    if (g_boot_mode != FASTBOOT_MODE) {
> +        ulong begin = get_timer(0);
> +        
> +        while (get_timer(begin)<50) {
> +            if (mtk_detect_key(MT_CAMERA_KEY)) {
> +                g_boot_mode = FASTBOOT_MODE;
> +                dprintf(INFO,"[FASTBOOT]Key Detect\n");
> +            }
> +        }
> +    }
> +#endif
>  #ifdef LK_PROFILING
>  	dprintf(CRITICAL, "[PROFILE] ------- boot mode select takes %d ms -------- \n", (int)get_timer(time_boot_mode));
>  #endif
> ```

Before 3.10 kernels mtk kernels usually shipped with lk inherited like in case of wiko mobiles which is opensource from quite sometime

p0lishit, 2019-02-16:
@TheInfected 

Porting is not possible for me, i'm not expert/developer :cry:

TheInfected, 2019-02-16:
> @TheInfected
> 
> Porting is not possible for me, i'm not an expert/developer

Can you send me lk source of same platform?
I'll try to implement this there and will reverse and see how these instructions are preserved by the compiler and will act on your stock lk accordingly

p0lishit, 2019-02-16:
I dont own mt6765 platform code
thus code were taken from mt6769 lk source

TheInfected, 2019-02-16:
Give me link

p0lishit, 2019-02-16:
@TheInfected 
[MT6769](https://github.com/gemian/gemini-lk)

TheInfected, 2019-02-16:
> @TheInfected
> [MT6769](https://github.com/gemian/gemini-lk)

This lk will not work.
I have some generic alsps bootloader but there can be two solutions
1. Reverse whole bootloader and fix every required functions.
2. Hex edit the binary ( which seems easy but actually it's even harder than approach above ).

closed 2019-02-16 by ivan19871002