MiCode/Xiaomi_Kernel_OpenSource #1032: Riva Oreo Kernel Source

sunnyraj84348, 2019-02-16:
As riva already got stable oreo update

Sp when kernel source?

Ayushraj21, 2019-02-16:
Release kernel source before sunny die ;__;

LordShenron, 2019-02-17:
Xiaomi is slow as always

LinuxPanda, 2019-02-20:
Please release rosy kernel source first.

luk1337, 2019-02-20:
@LinuxPanda what about they won't release either?

LinuxPanda, 2019-02-20:
> @LinuxPanda what about they won't release either?

I'm more afraid of them squashing riva & rosy commits together and call it a day.

rama982, 2019-02-20:
@LinuxPanda surely they will do it and sleep

LordShenron, 2019-02-20:
> 
> I'm more afraid of them squashing riva & rosy commits together and call it a day.

Thats what they gonna do xd

sunnyraj84348, 2019-02-20:
Hope they will not squash it with redmi 7A

Enzaik, 2019-02-22:
I'm kindly asking for kernel source as well, I'm a kernel enthusiast and would appreciate a lot having it!

LordShenron, 2019-02-23:
> I'm kindly asking for kernel source as well, I'm a kernel development enthusiast and would appreciate a lot having it!

Pro Enzaik sur hack xiaomeme servers for kernel source :3

rama982, 2019-02-23:
It will released, just need some patient 🙄🙏

NaAnBcEhEuL9A8K7, 2019-03-02:
Waiting for release patiently😊

NihalShafi, 2019-03-02:
Waiting for Oreo kernel source and make a custom kernel for all ROMs also miui o

MrYacha, 2019-06-28:
wen kranul xd