MiCode/Xiaomi_Kernel_OpenSource #1045: mi9内核在vmlinux阶段出现警告：EXPORT symbol "gsi_write_clannel_scratch" [vmlinux] version generation failed...

sekaiacg, 2019-02-23:
编译内核使用的教程来自：
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone
我尝试了很多gcc版本，都有这个错误，区别是7.4，8.2.1编译终止。
同时，在Module.symvers文中，gsi_write_clannel_scratch函数符号表为空：
`0x00000000	gsi_write_channel_scratch	vmlinux	EXPORT_SYMBOL`

sekaiacg, 2019-02-23:
已知，没影响...c

closed 2019-02-23 by sekaiacg

nakixii, 2020-08-14:
scripts/mod/modpost.c L688注释掉

dirt2022, 2024-02-08:
> scripts/mod/modpost.c L688注释掉

The linker will not link correctly.So how can I fix it?:
```
ld.lld: error: relocation R_AARCH64_ABS32 cannot be used against symbol '__crc_gsi_write_channel_scratch'; recompile with -fPIC                                                                                     >>> defined in built-in.a(drivers/platform/msm/gsi/gsi.o)
```

YumeMichi, 2024-03-06:
> Hi having same issue with pipa-t-oss kernel

pick those 6 commits
https://github.com/LineageOS/android_kernel_qcom_sm8250/commits/lineage-20/drivers/platform/msm/gsi

sekaiacg, 2024-03-06:
> Hi having same issue with pipa-t-oss kernel

https://github.com/libxzr/android_kernel_oneplus_sm8250/commit/ddb65cfb44b4bdf52690ed06f71c9ffc51a34014