MiCode/Xiaomi_Kernel_OpenSource #1046: 编译出错

DroidKali, 2019-02-26:
/media/root/E8D49AD8223FB1C3/Xiaomi_Kernel_OpenSource/drivers/soc/qcom/subsystem_restart.c:1561:19: error: 'WT_SUBSYSTEM_RESTART_LEVEL' undeclared (first use in this function)
/media/root/E8D49AD8223FB1C3/Xiaomi_Kernel_OpenSource/drivers/soc/qcom/subsystem_restart.c:1561:19: note: each undeclared identifier is reported only once for each function it appears in

机型:Redmi 4X
编译Host: Ubuntu 18.04
gcc版本:Linaro gcc 4.9

ivan19871002, 2019-02-26:
删除这行即可，这行是为了debug   从外部配置的

DroidKali, 2019-02-26:
ok，我再试试

DroidKali, 2019-02-27:
依然报错
../drivers/soc/qcom/subsystem_restart.c: In function ‘subsys_register’:
../drivers/soc/qcom/subsystem_restart.c:1546:6: warning: unused variable ‘ret’ [-Wunused-variable]
error, forbidden warning: subsystem_restart.c:1546
../scripts/Makefile.build:257: recipe for target 'drivers/soc/qcom/subsystem_restart.o' failed
make[4]: *** [drivers/soc/qcom/subsystem_restart.o] Error 1
make[4]: *** 正在等待未完成的任务....

DroidKali, 2019-02-27:
你们能把源码改好后再上传吗？

DroidKali, 2019-02-27:
> 删除这行即可，这行是为了debug 从外部配置的

依然报错
../drivers/soc/qcom/subsystem_restart.c: In function ‘subsys_register’:
../drivers/soc/qcom/subsystem_restart.c:1546:6: warning: unused variable ‘ret’ [-Wunused-variable]
error, forbidden warning: subsystem_restart.c:1546
../scripts/Makefile.build:257: recipe for target 'drivers/soc/qcom/subsystem_restart.o' failed
make[4]: *** [drivers/soc/qcom/subsystem_restart.o] Error 1
make[4]: *** 正在等待未完成的任务....
你们能修改后上传吗？

DroidKali, 2019-02-27:
> 删除这行即可，这行是为了debug 从外部配置的

麻烦给个话啊

ivan19871002, 2019-03-01:
改好了，你自己去看新的提交吧。看过去你是基本不会改代码的同学。

closed 2019-03-01 by ivan19871002