MiCode/Xiaomi_Kernel_OpenSource #1048: Redmi Note 7 Pro Kernel Source

Haseo97, 2019-03-01:
Would be nice if released more faster like Mi 9 😀

sunnyraj84348, 2019-03-03:
This is not flagship device ;(

merothh, 2019-03-03:
> This is not a flagship device ;(

so what. that doesnt make it any different

ivan19871002, 2019-03-04:
Has it been sold yet？

rupansh, 2019-03-09:
> Has it been sold yet？

gib msm8937 little kernel much thank

rama982, 2019-03-12:
They will release it with unified kernel for other device, Tulip, Wayne, and Whyred. Beat me.

emman-bautista, 2019-03-13:
Mi9 Kernel was release first? I can see how top secret this note 7 kernel is. Which one is more stable to release, note 7 or mi 9? I can image the age of the device and the internal developer's attentions are pretty useless.

rama982, 2019-03-13:
what this, redmi note 7 lavender with sdm660?

Hasaber8, 2019-03-13:
> what this, redmi note 7 lavender with sdm660?

no, Redmi Note 7 Pro (India) with sdm675

rashedsahaji, 2019-03-17:
Please release

Russiarain, 2019-03-18:
发布会开完啦，内核源码快快公布吧。

Haseo97, 2019-03-19:
Up

Russiarain, 2019-03-19:
lavender 's kernel source has been released...it seems that note 7 and note 7 pro share the same one.

Haseo97, 2019-03-19:
> lavender 's kernel source has been released...it seems that note 7 and note 7 pro share the same one.

No

rama982, 2019-03-19:
@Russiarain nope, 7 pro use sdm675 which use kernel 4.9
Note 7 is still on sdm660 with kernel 4.4

Haseo97, 2019-03-19:
> @Russiarain nope, 7 pro use sdm675 which use kernel 4.9
> Note 7 is still on sdm660 with kernel 4.4

sm6150 uses 4.14 kernel (sd675)

rama982, 2019-03-19:
@Haseo97 oh yes, just know about it.

zixijian, 2019-03-20:
It’s not so good to call lavender now.lilac looks good.

Scissordragonboy, 2019-03-20:
> > @Russiarain nope, 7 pro use sdm675 which use kernel 4.9
> > Note 7 is still on sdm660 with kernel 4.4
> 
> sm6150 uses 4.14 kernel (sd675)

When can we expect note 7 pro kernel source released? As I have seen you did up kernel source for Redmi note 7 ...

Haseo97, 2019-03-20:
> > > @Russiarain nope, 7 pro use sdm675 which use kernel 4.9
> > > Note 7 is still on sdm660 with kernel 4.4
> > 
> > 
> > sm6150 uses 4.14 kernel (sd675)
> 
> When can we expect note 7 pro kernel source released? As I have seen you did up kernel source for Redmi note 7 ...

Need to wait for Xiaomi to release

Scissordragonboy, 2019-03-20:
> > > > @Russiarain nope, 7 pro use sdm675 which use kernel 4.9
> > > > Note 7 is still on sdm660 with kernel 4.4
> > > 
> > > 
> > > sm6150 uses 4.14 kernel (sd675)
> > 
> > 
> > When can we expect note 7 pro kernel source released? As I have seen you did up kernel source for Redmi note 7 ...
> 
> Need to wait for Xiaomi to release

If you got Redmi note 7. Bring up for vendor tree for Redmi note 7.

Haseo97, 2019-03-20:
> If you got Redmi note 7. Bring up for vendor tree for Redmi note 7.

Not interested in ROM bring-up or building, need kernel only.

Scissordragonboy, 2019-03-20:
> Not interested in ROM bring-up or building, need kernel only.

I want to build ROM for it. But I don't have this device. So that's why I can't bring up vendor blobs without having device. If you got an note 7. Then you can pull them from your device

rama982, 2019-03-20:
@Scissordragonboy you mean dump rom like this
https://github.com/TadiT7/xiaomi_violet_dump

closed 2019-03-22 by Haseo97