MiCode/Xiaomi_Kernel_OpenSource #1067: Redmi 7 (onclite) Kernel Source

rama982, 2019-03-29:
Hope it released asap too,

noiver, 2019-03-30:
Hope it released asap too,

laststandrighthere, 2019-03-30:
Waiting

Kenexed, 2019-04-20:
Developers i think you need too release the kernel source immediately because there are many developers are waiting for it and the phone itself has become popular due to its specs and low price and many people wanted to root their redmi 7 (onclite) i hope you release it asap

mylove90, 2019-04-26:
Please release the source code of the kernel

bju2000, 2019-04-30:
Moto g7 kernel https://github.com/Alberto97/android_kernel_motorola_sdm632

Motorola has already made public.

mylove90, 2019-05-03:
> Moto g7 kernel https://github.com/Alberto97/android_kernel_motorola_sdm632
> 
> Motorola has already made public.

how about extracting kernel configuration from boot.img and use it with moto g7 kernel sources to compile the kernel for the Redmi 7 ?

mylove90, 2019-05-03:
> > Moto g7 kernel https://github.com/Alberto97/android_kernel_motorola_sdm632
> > Motorola has already made public.
> 
> how about extracting kernel configuration from boot.img and use it with moto g7 kernel sources to compile the kernel for the Redmi 7 ?

I was able to extract the configuration 
Here we go : [https://pastebin.com/tNPphPR7](url)

rama982, 2019-05-04:
@mylove90 thats not simply, there is some specific driver changes and dts configuration which is not found on other device kernel source.

laststandrighthere, 2019-05-04:
> > Moto g7 kernel https://github.com/Alberto97/android_kernel_motorola_sdm632
> > Motorola has already made public.
> 
> how about extracting kernel configuration from boot.img and use it with moto g7 kernel sources to compile the kernel for the Redmi 7 ?

excuse me but wtf

mylove90, 2019-05-04:
Excuse me but i am still learning :D

Z5X67280, 2019-05-04:
@mylove90 excuce me but how can u get the dts and driver (etc) stuff from kernel 🤔 😂

rama982, 2019-05-27:
Released: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/onc-p-oss

closed 2019-05-27 by rama982

vvr3ddy, 2020-02-27:
> Please release the source code of the kernel

How bout bigNO

mylove90, 2020-03-17:
> > > Moto g7 kernel https://github.com/Alberto97/android_kernel_motorola_sdm632
> > > Motorola has already made public.
> > 
> > 
> > how about extracting kernel configuration from boot.img and use it with moto g7 kernel sources to compile the kernel for the Redmi 7 ?
> 
> excuse me but wtf

I remember that day now :D

tamim8, 2020-06-23:
Please release the source code of the kernel