MiCode/Xiaomi_Kernel_OpenSource #1070: Release RN5 (Whyred) Pie sources

Kugumin, 2019-03-31:
So, title speaks for itself. Update them to make them compatible with Pie firmware, as without it's support kernel won't boot in system

PrimoDev23, 2019-03-31:
So there is a way to make kernel booting, but frequencies are locked, many bugs are there and and and... Custom Kernel devs need the sources to give their users the best feeling and let them enjoy their work. Hope to get new pie sources soon!

zakiy99, 2019-04-02:
gib pie kernel source

GuaiYiHu, 2019-04-10:
This does work.
https://github.com/GuaiYiHu/android_kernel_xiaomi_whyred/commit/b2647f1433f94f7fd913d908aead108526586cd2

PrimoDev23, 2019-04-12:
Problem is that this only works for pie. It is not booting on Oreo anymore after this

GuaiYiHu, 2019-04-12:
Good, why not take nougat into consideration?

rama982, 2019-04-12:
Pie kernel changes will never boot on Oreo, same case on tulip

PrimoDev23, 2019-04-12:
And exactly this isn't right. There is a decompiled dts version that works for Oreo aswell as for pie firmwares (for whyred its there, idk about tulip)
I wouldn't have asked if this statement would be true. The only problem (and the fact why I don't like this decompiled dts version) is that we can't easily add includes. We can't even add some nodes bc they are always giving errors. I am not talking about every nodes, just about a few... So hope anyone will find a solution for this

rama982, 2019-04-12:
The solution is when Xiaomi push pie kernel source public.
For now we only can settle with that temporary solution i think

rcstar6696, 2019-04-12:
On wayne all good no issues with pie a2 kernel base on oreo or pie custom rom

shekhawat2, 2019-05-06:
> This does work.
> [GuaiYiHu/android_kernel_xiaomi_whyred@b2647f1](https://github.com/GuaiYiHu/android_kernel_xiaomi_whyred/commit/b2647f1433f94f7fd913d908aead108526586cd2)

this works perfectly. but dt2w gesture is not working on some devices. and fukin drains battery when enabled.
we need phix either from xiaomi or from you.😁

Kugumin, 2019-05-13:
Sources were released!

closed 2019-05-13 by Kugumin