MiCode/Xiaomi_Kernel_OpenSource #1074: mi9 camera not working, miui 9.4.8

sekaiacg, 2019-04-09:
`04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521a000] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521a300] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521a600] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521a900] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521ac00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521af00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521b200] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521b500] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521b800] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521bb00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521be00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521c100] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521c400] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521c700] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521ca00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790521cd00] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790454e000] Warning: Release called on an un-backed buffer
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1
04-09 10:16:58.074 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffer.cpp:800 Release(): 0 == referenceCount
04-09 10:16:58.074 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATOfflinePreview_ChiNodeWrapper1_InputPort0_ ][0x790454e300] Warning: Release called on an un-backed buffer
[... cut ...]
04-09 10:16:58.087 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATJpegEncode_IPE4_InputPort0_Full][0x791a3b8d00] Warning: Release called on an un-backed buffer
04-09 10:16:58.087 +0000   729  1664 W CamX    : [ERROR][Always ] camximagebuffer.cpp:890 Release(): [SATJpegEncode_IPE4_InputPort0_Full][0x791a1bb100] Warning: Release called on an un-backed buffer
04-09 10:16:58.088 +0000   729  1664 E CamX    : [ERROR][Assert ] camximagebuffermanager.cpp:828 FreeBufferFromList(): pBuffer->ReferenceCount 1`
emmmmmm

xiangfeidexiaohuo, 2019-04-09:
Expect to update source code

K-996, 2019-04-09:
Expect to update source code

ivan19871002, 2019-04-11:
Than you for your feedback.

We will check it  internally

sekaiacg, 2019-04-16:
```
04-09 10:17:51.892 +0000  2299  2621 I NotifierManager: onRegister: miui.maml.NotifierManager$MultiBroadcastNotifier@19e154c
04-09 10:17:52.030 +0000   968  6869 E Camera3-Device: Camera 120: waitUntilDrainedLocked: Error waiting for HAL to drain: Connection timed out (-110)
04-09 10:17:52.157 +0000   968  6869 D CameraTraces: Process trace saved. Use dumpsys media.camera to view.
04-09 10:17:52.157 +0000   968  6869 E CameraDeviceClient: detachDevice: waitUntilDrained failed with code 0xffffff92
04-09 10:17:52.157 +0000   968  6869 I Camera3-Device: disconnect: E
04-09 10:17:52.157 +0000   968  6869 E Camera3-Device: Camera 120: disconnect: Shutting down in an error state
```

ivan19871002, 2019-04-27:
4月初相机有大改，已经安排开源计划了。

closed 2019-06-13 by sekaiacg