MiCode/Xiaomi_Kernel_OpenSource #1076: Redmi 7 (onclite)

Kenexed, 2019-04-20:
Hey developers can you please release the kernel source code of redmi 7 many people are waiting for it

betawi, 2019-04-26:
Any eta on this??

mylove90, 2019-05-03:
Still waiting for this

sntdevco, 2019-05-16:
> Any eta on this??

Probably ~23rd May (GMT -8)

Haseo97, 2019-05-27:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/onc-p-oss

kairusds, 2019-05-27:
please close the issue

mylove90, 2019-06-21:
Anyone tried to compile this ?

i got that error

ERROR (phandle_references): Reference to non-existent node or label "typec_ssmux_config"

ERROR: Input tree has errors, aborting (use -f to force output)
scripts/Makefile.lib:319: recipe for target 'arch/arm64/boot/dts/qcom/apq8053-lite-dragon-v1.0.dtb' failed
make[3]: *** [arch/arm64/boot/dts/qcom/apq8053-lite-dragon-v1.0.dtb] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC      drivers/char/diag/diag_usb.o
  CC      drivers/char/diag/diagmem.o
  CC      kernel/sched/idle.o
../scripts/Makefile.build:653: recipe for target 'arch/arm64/boot/dts/qcom' failed
make[2]: *** [arch/arm64/boot/dts/qcom] Error 2
arch/arm64/Makefile:161: recipe for target 'dtbs' failed
make[1]: *** [dtbs] Error 2
make[1]: *** Waiting for unfinished jobs....

Any help on this ?

Dhina17, 2019-07-03:
Please release stable source for onclite as soon as possible...

mylove90, 2019-07-03:
Just for the info...i was able to resolve the above mentioned error but i got another couple of error that i override it and got the sources to compile but it didn't boot on the device and got stuck in the Redmi logo....this is not normal...manufactures release kernel sources the just work out of the box

rama982, 2019-07-03:
@mylove90 remove avb flag on dtsi file

mylove90, 2019-07-03:
@rama982 i will try that when i get back home for sure

Zer0nite, 2019-12-19:
@mylove90 I have the same problem. Did you fix it?

Zer0nite, 2019-12-19:
@rama982 @mylove90 I'm getting this error too

`  DTC     arch/arm64/boot/dts/qcom/apq8053-lite-dragon-v1.0.dtb`
`ERROR (phandle_references): Reference to non-existent node or label "typec_ssmux_config"`
`ERROR: Input tree has errors, aborting (use -f to force output)`
`make[2]: *** [scripts/Makefile.lib:319: arch/arm64/boot/dts/qcom/apq8053-lite-dragon-v1.0.dtb] Error 2`
`make[1]: *** [scripts/Makefile.build:653: arch/arm64/boot/dts/qcom] Error 2`
`make: *** [arch/arm64/Makefile:161: dtbs] Error 2`

Changed this line in: arch/arm64/boot/dts/qcom/msm8953.dtsi
`fsmgr_flags = "wait,avb";` to `fsmgr_flags = "wait";`
But the problem exists.

luk1337, 2019-12-19:
[image]
oh wow i wonder why that didn't change anything...

Zer0nite, 2019-12-19:
Idea?

NCLnclNCL, 2025-10-07:
> [@rama982](https://github.com/rama982) i will try that when i get back home for sure

How to fix build