MiCode/Xiaomi_Kernel_OpenSource #1077: mi6|mix3会不会有源码更新？

sekaiacg, 2019-04-25:
如题，不更新就直说了，告诉大家不要等了。
As the title, if you don't update, just say it, tell everyone not to wait.

ivan19871002, 2019-04-26:
米6和MIX3 是不同的平台 不同的代码，你放一起是为了说明啥？？

另外 目前的开源代码不匹配当前的ROM吗？有哪一项不能正常工作吗？

ivan19871002, 2019-04-26:
版本有大更新或者内部有比较重要的功能更新的话，会再次开源的，或者如果你们发现哪些功能已经无法使用，我们没有及时开源的，可以通过issue 告诉我们。
目前你这样开issue 我不知道到底需要更新啥啊

sekaiacg, 2019-04-26:
> 米6和MIX3 是不同的平台 不同的代码，你放一起是为了说明啥？？
> 
> 另外 目前的开源代码不匹配当前的ROM吗？有哪一项不能正常工作吗？

为了方便我把mi6与mix3放一起了。
mi6属于安卓版本更新，需要再次开源，不知道要等多久。
个人感觉mi6还能再战2年，待机方面比mi9强不止一点。希望不要放弃。

sekaiacg, 2019-04-26:
> 版本有大更新或者内部有比较重要的功能更新的话，会再次开源的，或者如果你们发现哪些功能已经无法使用，我们没有及时开源的，可以通过issue 告诉我们。
> 目前你这样开issue 我不知道到底需要更新啥啊
至于mix3，感觉被放弃一样，传感器不工作，状态栏网速也不工作
有相关的issue：https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/1059

sekaiacg, 2019-04-26:
> 版本有大更新或者内部有比较重要的功能更新的话，会再次开源的，或者如果你们发现哪些功能已经无法使用，我们没有及时开源的，可以通过issue 告诉我们。
> 目前你这样开issue 我不知道到底需要更新啥啊

至于mix3，感觉被放弃一样，传感器不工作，状态栏网速也不工作
有相关的issue：#1059
另外：mi9的待遇会是这样吗？

xiangfeidexiaohuo, 2019-04-26:
> 版本有大更新或者内部有比较重要的功能更新的话，会再次开源的，或者如果你们发现哪些功能已经无法使用，我们没有及时开源的，可以通过issue 告诉我们。
> 目前你这样开issue 我不知道到底需要更新啥啊

米6 更新安卓P 肯定需要再次开源，目前的源码 NFC是坏的，米6只要更新下去，就可以成为小米2s那样的神器，比最新的小米9强的不止一点。

MIX3，应该从3月份的开发开始，距离感应器就挂了，之前有人反馈过。MIX3一直小毛病不断，曾经给秋大说过，不知道是没收到，还是不重视，反正2个月了，没任何回应，这样的情况，难免不会让人觉得官方已经放弃了MIX3

ivan19871002, 2019-04-27:
sagit 更新p 稳定版之后会安排再次开源，mix3 会内部再查一下

sekaiacg, 2019-07-11:
sagit-p-oss已发布

closed 2019-07-11 by sekaiacg

wode2016501, 2019-10-01:
小米6 怎么选defconf  6+64g

soenggam, 2019-10-01:
@wode2016501 请您不要在多个issues发同样的问题，好么？麻烦您自己多读读代码看看提交记录。

wode2016501, 2019-10-02:
我百度也 找不到办法的  百度上说AndroidBoard.mk  问题这个源码没有呀 都不知道怎么选defconfig