MiCode/Xiaomi_Kernel_OpenSource #1079: Fake Open Source Support

scafroglia93, 2019-05-01:
- Lack of kernel side updates. 
There is no maintenance by this vaunted manufacturer within the released source. 
It's a snapshot thrown at them for fools

- We want to talk about the qualcomm base used for the development of the system which does not receive any updates

These would be the famous optimized phones?

Or maybe they are loved for the decidedly lower cost since there is no real support from the producer; producer that every 3x2 is exposed to security problems within its system because, we all know that there is a real interest in the data of the user so loved by Xiaomi for his love policies towards Big Data
There are so many projects that deserve more attention but I admit that the marketing used to raise the buyer towards a false community has a positive outcome.

ryan-andri, 2019-05-01:
lol

scafroglia93, 2019-05-01:
It does not seem correct considering that the majority of those who support this company (not receiving support) exploit material released and constantly updated

I don't think Kernel.org has the same methodology of working with snapshots thrown at random without any support.

We see that the cost of the product is able to overcome every type of ethics

soenggam, 2019-05-01:
what's the problem?i don't realize.

closed 2019-05-03 by ivan19871002