MiCode/Xiaomi_Kernel_OpenSource #1081: 小米8青春版 platina ADSP出现问题

rsyhan, 2019-05-07:
目前用的是给出的MAX3, MI 8 Lite （nitrogen-p-oss分支），不做任何直接编译出来刷入MIUI 9.4.25中，查看日志发现ADSP炸了。
在无内核源码时，使用 MIUI 9.2.* 的预编译内核和 vendor 时，无此问题。自从把 vendor 和 Prebuilt Kernel 升级至 9.3.* 后台存开始出现 ADSP 爆炸。
此外，在类原生中必须要使用 MIUI 9.2.* frimware，否则audio直接挂掉。使用 MIUI 最新 Prebuilt Kernel 时无此问题。
附件为目前正在使用的 MIUI 系统版本和 ADSP 的 crash log
[image]
[image]

ivan19871002, 2019-05-07:
没明白你要表达的意思，我不管类原生和第三方所有ROM
你帮忙把在MIUI版本上 会出问题的组合都整理一下即可

rsyhan, 2019-05-07:
目前直接用nitrogen p oss分支的源码编译出来刷入MIUI 9.4.25，ADSP炸了
简单说就这一个问题
crash log在上面

ivan19871002, 2019-05-07:
9.4.25之前的版本呢

rsyhan, 2019-05-07:
自从开源之后的MIUI版本都炸了应该
MIUI 9.3.*测试过一次是炸的
MIUI 9.4.*测试过两个版本是炸的
具体日期记得不的了

ivan19871002, 2019-05-07:
log 用https://pastebin.com 发出来

另外 我看你这个简单的log 跟我认为的adsp挂是不一样的 挂， 你确定声音是没有的吗

rsyhan, 2019-05-07:
抱歉这两天学校组织旅游信号不好编译测试较慢。
https://pastebin.com/ZWndwG6m
经过测试 audio sensor adsp 目前这三个炸了 直接p oss未改动
后期还有什么炸不清楚
我在类原生开发中采取的方案是用 MIUI 9.2.*的 firmware ，除了 adsp 其他都没问题了

ivan19871002, 2019-05-07:
log 不够，给我一份284吧

rsyhan, 2019-05-07:
http://dl-ht2.rsyhan.ml/dl/platina-kernel-oss/ivan.log
完整的存在这里了 从开机开始的 请速度下载查看～
您说的284是什么我不理解，还需要吗？

ivan19871002, 2019-05-07:
我double check 了一下，开源的代码是基于稳定版的，请使用稳定版的vendor 和匹配的kernel。不匹配开发版

closed 2019-05-07 by ivan19871002

ivan19871002, 2019-05-07:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/dc8e417a8d54d8c0893f19b97fb448d2a72b058d

这条修改，开发版上 改了adsp 挂载名称，需要配合对应开发版的 firmware

rsyhan, 2019-05-07:
@ivan19871002 谢谢男神，辛苦您了