MiCode/Xiaomi_Kernel_OpenSource #1082: We need source code for Cepheus google q beta kernel please.

randomblame, 2019-05-07:
We need source code for Cepheus google q beta kernel please. Released binary is at 4.14.105 the provided source is much older and non functional on the q beta. Please provide the source code per the gpl

scafroglia93, 2019-05-09:
LOL

randomblame, 2019-05-09:
Something funny scafroglia? If an entity releases a compiled version of the Linux kernel they MUST also release the corresponding source code. This is how Linux works. Hundreds of thousands of people have put millions of hours of work into it and all that anyone asks in return for this free software is compliance with the terms under which it was made available. Especially considering the circumstances of this release being in partnership with Google I expect nothing less from xiaomi than compliance with the gpl.

msdx321, 2019-05-09:
> Something funny scafroglia? If an entity releases a compiled version of the Linux kernel they MUST also release the corresponding source code. This is how Linux works. Hundreds of thousands of [...]

He's on your side I would say. [Check this](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/1079)

randomblame, 2019-05-09:
I just don't want to see anyone scoff at the idea of honoring gpl requests like it's a joke it perpetuates the problem. :)

sekaiacg, 2019-05-10:
I don't have any expectations for the source code with Android Q, because it is not released in a short time and it is difficult to get a quick update.
So far, the open source code has not been updated.

scafroglia93, 2019-05-11:
Needless to say, you are customers only for the price of the device, not to mention that every time you repeat the same story with the kernel releases you know very well that Xiaomi is not interested in this.

Not for nothing are released milestones without maintenance

There are as many OEMs that release so, change and you are okay unless you are so poor that you have to drop your pants by buying Xiaomi products

randomblame, 2019-05-11:
I a a customer because this was the first device using this soc available and because I had not heard anything negative about this company. Why are you even on GitHub?

YumeMichi, 2019-05-11:
There is only one Google! You can't ask every company to open source their code every day like Google.

randomblame, 2019-05-11:
Sure you can that's how open source software works.

msdx321, 2019-05-11:
@YumeMichi To be fair, Google update their devices' kernel sources once per month along with the security patch. But indeed, they're definitely the first choice if you want to get the latest sources for the device you buy.

msdx321, 2019-05-11:
> Needless to say, you are customers only for the price of the device, not to mention that every time you repeat the same story with the kernel releases you know very well that Xiaomi is not [...]

Dropping pants to but is going to far man. I was and probably still is a victim for Xiaomi's lame sources. Has been bothered by wrong defconfig, unmatched drivers and missing patches a lot before. But those aren't the reasons you can making fun of her just because she's doing the right thing here.

randomblame, 2019-05-11:
It's ridiculous that a simple gpl request on GitHub results in such a hateful and misogynistic response. There is no need for anyone else to reply to this ticket except xiaomi representatives. I don't want to hear apologizers, recommendations for other products or any other nonsense. Can we please keep GitHub a professional environment?

YumeMichi, 2019-05-11:
Yes, you are right! Forgive my ignorance. @randomblame @msdx321

FaserF, 2019-05-17:
+1 for releasing the kernel source @Xiaomi