MiCode/Xiaomi_Kernel_OpenSource #1084: Tiare Broken Source

sunnyraj84348, 2019-05-21:
Even it can't get compiled successfully lol

rama982, 2019-05-21:
gib log, maybe xiaomi dev will fix

0xXA, 2019-05-26:
> Even it can't get compiled successfully lol

[Watch](https://asciinema.org/a/NgBCkKUZOWmGPV5iH6avxMIL1)

sunnyraj84348, 2019-05-26:
> > Even it can't get compiled successfully lol
> 
> [Watch](https://asciinema.org/a/NgBCkKUZOWmGPV5iH6avxMIL1)

Oh man , why it has riva dtb

0xXA, 2019-05-26:
> Oh man , why it has riva dtb
Those two have same boards that's why some pll and memory reasons should differ or possibly xiaomi used same code of Riva and ported to this device **redmi go**

sunnyraj84348, 2019-05-26:
> > Oh man , why it has riva dtb
> 
> Those two have same boards,pll and memory should differ or possibly xiaomi used same code of Riva and ported to this device **redmi go**

Oh, I thought they released riva oreo source with tiare 
Thanks.

closed 2019-05-26 by sunnyraj84348