MiCode/Xiaomi_Kernel_OpenSource #1092: [grus-p-oss] goodix TS - FOD issues on miui 9.6.5

okitavera, 2019-06-13:
It works before on 9.5.25, but on the 9.6.5 it can't read any events on the FOD's anymore.

Also many things that I saw on the 9.6.5's kernel that doesn't present yet on this tree,
for example : 
- rwsem prio aware enhancement
- hwconf_manager
- palmsensor and gamemode touchfeature

any plans for source updates ? thanks.

okitavera, 2019-06-15:
after comparing dmesg, I saw that it has a new `xiaomitouch_register_modedata()` and `gtp_init_touchmode_data()`, and the `goodix_gt9886_cfg_f2.bin` has a different esd register and bumped version.

Gah0, 2019-06-15:
before miui 9.6.5 
`[    0.966609] [GTP-INF][goodix_parse_cfg_bin:153] esd reg:303f`

after miui 9.6.5
```
[    0.976325] [GTP-INF][goodix_parse_cfg_bin:153] esd reg:30f3
[    1.178402] [GTP-INF][goodix_check_update:307] Firmware needs to be updated
```

okitavera, 2019-06-26:
still no updates huh

closed 2019-07-03 by okitavera