MiCode/Xiaomi_Kernel_OpenSource #1104: cepheus-p-oss 编译报错

suasm, 2019-07-03:
Ubuntu19.04  手机mi9 跟着https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone这边做的
`git clone --depth=1 https://github.com/MiCode/Xiaomi_Kernel_OpenSource.git -b cepheus-p-oss cepheus-p-oss
cd cepheus-p-oss
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9 toolchain

下载snapdragon-llvm-compiler-android-linux64-609.tar.gz

tar vxzf snapdragon-llvm-compiler-android-linux64-609.tar.gz
mkdir out
export ARCH=arm64
export SUBARCH=arm64
export DTC_EXT=dtc
export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-android-Set CONFIG_BUILD_ARM64_DT_OVERLAY=y
运行如下make 报错：
make O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- cepheus_user_defconfig
报错情况：
make[1]: Entering directory '/home/suroot/bin/cepheus-p-oss/out'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  GEN     ./Makefile
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
make[1]: Leaving directory '/home/suroot/bin/cepheus-p-oss/out'
`
如何解决？

YumeMichi, 2019-07-03:
好像没看到哪里写着报错

ivan19871002, 2019-07-04:
没有看到哪里报错了

suasm, 2019-07-04:
感谢你的回复！应该是这个make命令报错;make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log  第一次提问手贱了弄错了 我后来用Ubuntu18.04编译成功 我想是19.04配的编译环境不全引起的

suasm, 2019-07-04:
执行这个make：make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log报错为：
make[1]: Leaving directory '/home/suroot/bin/cepheus-p-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2

YumeMichi, 2019-07-04:
所以贴了这么多还是没把错误的地方贴出来=。=

sekaiacg, 2019-07-11:
这只是告诉你有子模块编译失败，具体是什么未知。
建议将kernel.log上传...

suasm, 2019-07-11:
我怀疑是dtc版本的问题，我去查了一下源码中的 scripts/dtc/version_gen.h 提示版本为#define DTC_VERSION "DTC 1.4.4-g756ffc4f"
可是我安装了dtc1.4.7之后检查版本依然是1.4.0的版本，(sudo apt install ./device-tree-compiler_1.4.7-1_amd64.deb)这是为何？，还有我是否能够直接从源码中编译出一个dtc出来？

YumeMichi, 2019-07-11:
所以到底为什么不直接贴完整日志了 (((φ(◎ロ◎;)φ)))

suasm, 2019-07-11:
抱歉！是我的错。给你传附件，你帮看看吧。

sekaiacg, 2019-07-11:
@suasm 将这个去掉->export DTC_EXT=dtc。就会使用你安装的版本。
官方对dtc有要求：dtc must be from aosp source code（pie-release） Clang from qcom

suasm, 2019-07-11:
谢谢你提供的方法，我先去试试。

suasm, 2019-07-11:
我去掉export DTC_EXT=dt之后进行编译，结果报了很多的警告，日志发附件上了，能帮我再看看吗？

mrslezak, 2019-07-20:
I also have issues compiling.  What is the DTC version that is Pie to be installed rather than the one installed by apt install dtc?

sekaiacg, 2019-07-22:
https://android.googlesource.com/platform/external/dtc/

mrslezak, 2019-07-25:
I answered my own question you need to use DTC 1.4.6 or higher to get it to compile.  Now I am trying to figure out how to get the SDM845 Wifi and Audio drivers into the Vendor partition per the same guide since the Cepheus build does not have audio or wifi.

iesnekei, 2022-06-08:
> https://android.googlesource.com/platform/external/dtc/

哥们儿，俄罗斯小伙找你问问你会不会像别的手机做Xiaomi 12pro (zeus)的手机的系统的启动器？他们想弄一弄俄文系统，可是没有你的那个什么东西，他们不会。