MiCode/Xiaomi_Kernel_OpenSource #1108: 为什么“设置-我的设备-全部参数-内核版本”，该内核在github上找不到相应的提交

shenzhigang, 2019-07-15:
如图，我找了2个手机（max 2和mix 2）内核版本 1472e29 和 1968ad0 在github上都找不到相应都提交。
[image]

YumeMichi, 2019-07-15:
因为开源的肯定不一样啊 比如内部都是每个修改都有一个提交 但是开源把全部的修改都合成一个提交了

shenzhigang, 2019-07-15:
> 因为开源的肯定不一样啊 比如内部都是每个修改都有一个提交 但是开源把全部的修改都合成一个提交了

我想基于这个版本的内核，修改一个参数然后编译，我该如何找到和当前内核版本最近的这次开源的提交呢？

YumeMichi, 2019-07-15:
不可能的 小米几个月一年才开源一次

shenzhigang, 2019-07-15:
> 不可能的 小米几个月一年才开源一次

也就是说，现在开源的代码，是一年前的某个发布版本的内核

shenzhigang, 2019-07-16:
谢谢

closed 2019-07-16 by shenzhigang

Z5X67280, 2019-07-16:
小米的 Kernel Defconfig 默认启用了 CONFIG_LOCALVERSION_AUTO，打开这个选项之后 LOCALVERSION 就会带一个 -gXXXXXXX

虽然开出来的是稳定版内核但不一定是最新稳定版的内核，除非有重大更改否则不会更新内核（这里要夸一下一加，每一次更新之后都会更新一下源码）

原版的内核是一个一个 commit 的，和开出来的一下子 300000+ added 的 commit 不同，所以后面那串 SHA1 不一样

而且最好笑的一点……
小米给的都不是合并性提交，所以你想查单个单个 commit 都是查不到的

就酱 :-