MiCode/Xiaomi_Kernel_OpenSource #1109: README.md里Redmi 5plus的Base tag 有误？LA.UM.5.3-06310-89xx.0 这个是Android6.0.1

shenzhigang, 2019-07-16:
这是README文件里的截图
[image]

以下是android官方查到的LA.UM.5.3-06310-89xx.0 ，对应的是6.0.1
[image]

另外，tiffany-n-oss分支（Redmi 5plus和小米5x共用此内核？）编译之后Image.gz-dtb文件大小要比目前最新版本的红米5plus的线刷包里的kernel文件小3MB左右，刷入后也无法开机。是不是Redmi 5plus的内核放错了？

Z5X67280, 2019-07-16:
vince 第三方 Linux 4.9 内核开发者在此
vince 的 MIUI 最新版是 Android 8.1，tiffany-n-oss 下的是给旧版 MIUI 的 Android 7.x 用的，请换成 sakura-o-oss 下的内核源码
至于 LA.UM.5.3-XXXXX-89xx.0，确实是给 Android 7.x 用的

shenzhigang, 2019-07-17:
> markw

谢谢，我用了sakura-o-oss就可以了

closed 2019-07-17 by shenzhigang