MiCode/Xiaomi_Kernel_OpenSource #1110: raphael-p-oss编译错误

atomsand, 2019-07-18:
是从这里开始报错的
../scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: 没有那个文件或目录
 #include <openssl/bio.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
../scripts/sign-file.c:25:10: fatal error: openssl/opensslv.h: 没有那个文件或目录
 #include <openssl/opensslv.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.host:102: recipe for target 'scripts/extract-cert' failed
make[2]: *** [scripts/extract-cert] Error 1
make[2]: *** 正在等待未完成的任务....
scripts/Makefile.host:102: recipe for target 'scripts/sign-file' failed
make[2]: *** [scripts/sign-file] Error 1
  CHK     include/generated/timeconst.h
  CC      kernel/bounds.s
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/bounds.h
  CC      arch/arm64/kernel/asm-offsets.s
/home/andy/Atom-Kernel/raphael/Makefile:581: recipe for target 'scripts' failed
make[1]: *** [scripts] Error 2
make[1]: *** 正在等待未完成的任务....
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
make[1]: 离开目录“/home/andy/Atom-Kernel/raphael/out”
Makefile:146: recipe for target 'sub-make' failed

sekaiacg, 2019-07-19:
缺少openssl，根据你的环境安装却少的软件包。

atomsand, 2019-07-19:
已经安装了openssl,现在安装libssl-dev再试试

atomsand, 2019-07-19:
> 缺少openssl，根据你的环境安装却少的软件包。
问题解决了，谢谢dalao,学习了

atomsand, 2019-07-19:
又有新问题了
  DTC     arch/arm64/boot/dts/qcom/andromeda-sm8150-overlay.dtbo
Error: ../arch/arm64/boot/dts/qcom/andromeda-pinctrl.dtsi:3.1-6 syntax error
FATAL ERROR: Unable to parse input tree
make[3]: *** [scripts/Makefile.dtbo:24：arch/arm64/boot/dts/qcom/andromeda-sm8150-overlay.dtbo] 错误 1
make[2]: *** [../scripts/Makefile.build:678：arch/arm64/boot/dts/qcom] 错误 2
make[1]: *** [arch/arm64/Makefile:184：dtbs] 错误 2
make[1]: 离开目录“/home/andy/Atom-Kernel/raphael/out”
make: *** [Makefile:146：sub-make] 错误 2

ivan19871002, 2019-07-19:
看一下wiki吧，我已经明确说明了，需要用android P 版本里面编译出来的dtc

atomsand, 2019-07-19:
> 看一下wiki吧，我已经明确说明了，需要用android P 版本里面编译出来的dtc
wiki好像关le............................

wufixx, 2019-07-19:
try run this sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev libjpeg8-dev zlib1g-dev

atomsand, 2019-07-19:
> try run this sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev libjpeg8-dev zlib1g-dev

Thank you so much bro, it's so nice of you

wufixx, 2019-07-19:
> > try run this sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev libjpeg8-dev zlib1g-dev
> 
> Thank you so much bro, it's so nice of you

Yo

csrutil, 2019-07-20:
好像没有wiki。

PIPIPIG233666, 2019-07-20:
wiki昨天被哪个缺德的人删了

shandongtlb, 2019-07-22:
> 看一下wiki吧，我已经明确说明了，需要用android P 版本里面编译出来的dtc

可是wiki现在没了2333

csrutil, 2019-07-23:
页面缓存，https://webcache.googleusercontent.com/search?q=cache:QsyygrbQjbYJ:https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone+&cd=3&hl=en&ct=clnk&gl=us

closed 2019-07-26 by atomsand