MiCode/Xiaomi_Kernel_OpenSource #1111: 关于dipper-p-oss 分支问题

wilson404, 2019-07-19:
在高通[QAEP](https://wiki.codeaurora.org/xwiki/bin/QAEP/release) 上没有找到对应的tag LA.UM.7.3.r1-04500-sdm845.0
在[git://codeaurora.org/platform/manifest.git](https://source.codeaurora.org/quic/la/platform/manifest/tree/?h=release)的release 分支中找不到 LA.UM.7.3.r1-04500-sdm845.0.xml 文件 
请问下我应该如何使用这个内核源码来编译一整个aosp/caf

I can't find tag:LA.UM.7.3.r1-04500-sdm845.0 from [QAEP](https://wiki.codeaurora.org/xwiki/bin/QAEP/release)
How can I build a CAF for dipper with this tag.

YumeMichi, 2019-07-19:
这只是内核 要编译pure CAF你得自己解决

wilson404, 2019-07-19:
> 这只是内核 要编译pure CAF你得自己解决

所以是要自己从 LA.UM.7.3.r1-04500-sdm845.0 升级到 LA.UM.7.3.r1-05100-sdm845.0
然后我才可以编译pure CAF 对吗？

PIPIPIG233666, 2019-07-20:
你还要写dt

wilson404, 2019-07-20:
> 你还要写dt

dt 的意思是 设备树（device tree ）的意思吗？
（我这是不知者无畏吗，难度很大的样子，弃疗。。。

Gah0, 2019-08-05:
这里只是内核
你的845,caf标签在https://source.codeaurora.org/quic/la/kernel/msm-4.9/
对内核合并caf标签，解决冲突后，得到的就是CAF内核。