MiCode/Xiaomi_Kernel_OpenSource #1113: raphael-p-oss编译问题

csrutil, 2019-07-24:
```
root@kernel:~# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:  Ubuntu 18.04.2 LTS
Release:  18.04
Codename: bionic

sudo apt-get install git ccache automake flex lzop bison \
  gperf build-essential zip curl zlib1g-dev zlib1g-dev:i386 \
  g++-multilib python-networkx libxml2-utils bzip2 libbz2-dev \
  libbz2-1.0 libghc-bzlib-dev squashfs-tools pngcrush \
  schedtool dpkg-dev liblz4-tool make optipng maven libssl-dev \
  pwgen libswitch-perl policycoreutils minicom libxml-sax-base-perl \
  libxml-simple-perl bc libc6-dev-i386 lib32ncurses5-dev \
  x11proto-core-dev libx11-dev lib32z-dev libgl1-mesa-dev xsltproc unzip \
  mosh htop lsof screen -y

git clone --depth=1 https://github.com/MiCode/Xiaomi_Kernel_OpenSource.git -b $BRANCH $BRANCH

cd $BRANCH
mkdir out

git clone https://android.googlesource.com/platform/external/dtc -b pie-release
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9 toolchain
tar xvf /root/snapdragon-llvm-6.0.9-linux64.tar.gz -C .

export ARCH=arm64
export SUBARCH=arm64
export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-android-
export CONFIG_BUILD_ARM64_DT_OVERLAY=y

make O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- raphael_user_defconfig
make -j$(nproc) O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log

```

错误日志

```
  AR      drivers/usb/gadget/function/usb_f_mass_storage.o
  AR      drivers/usb/gadget/function/usb_f_fs.o
  AR      drivers/usb/gadget/function/usb_f_uac2.o
  AR      drivers/usb/gadget/function/usb_f_midi.o
  AR      drivers/usb/gadget/function/usb_f_hid.o
  AR      drivers/usb/gadget/function/usb_f_mtp.o
  AR      drivers/usb/gadget/function/usb_f_ptp.o
  AR      drivers/usb/gadget/function/usb_f_audio_source.o
  AR      drivers/usb/gadget/function/usb_f_accessory.o
  AR      drivers/usb/gadget/function/usb_f_diag.o
  AR      drivers/usb/gadget/function/usb_f_cdev.o
  AR      drivers/usb/gadget/function/usb_f_ccid.o
  AR      drivers/usb/gadget/function/usb_f_qdss.o
  AR      drivers/usb/gadget/function/usb_f_gsi.o
  AR      drivers/usb/gadget/function/built-in.o
  AR      drivers/usb/gadget/built-in.o
  AR      drivers/usb/built-in.o
  AR      drivers/built-in.o
make[1]: Leaving directory '/root/raphael-p-oss/out'
Makefile:146: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
```

okitavera, 2019-07-24:
The log is incomplete but if you sure it's about dtc issues,
You can use dtc from kernel.org instead, especially if you're doing a standalone compile.
```
$ git clone https://git.kernel.org/pub/scm/utils/dtc/dtc.git dtc-1.4.6 -b v1.4.6 
$ cd dtc-1.4.6/
$ make NO_PYTHON=1
```
then you can resume your kernel compile, don't forget to set the `DTC_EXT` before start compiling :
```
$ export DTC_EXT="path/to/dtc-1.4.6/dtc"
$ make blahblahblah....
```

csrutil, 2019-07-24:
Thank you for your replies. I will have a try :)

csrutil, 2019-07-24:
I just try, it does not work.

```
root@kernel:~# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:  Ubuntu 18.04.2 LTS
Release:  18.04
Codename: bionic

root@kernel:~# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04.1' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)

```

Here is the script

```
BRANCH=raphael-p-oss

sudo apt-get install git ccache automake flex lzop bison \
  gperf build-essential zip curl zlib1g-dev zlib1g-dev:i386 \
  g++-multilib python-networkx libxml2-utils bzip2 libbz2-dev \
  libbz2-1.0 libghc-bzlib-dev squashfs-tools pngcrush \
  schedtool dpkg-dev liblz4-tool make optipng maven libssl-dev \
  pwgen libswitch-perl policycoreutils minicom libxml-sax-base-perl \
  libxml-simple-perl bc libc6-dev-i386 lib32ncurses5-dev \
  x11proto-core-dev libx11-dev lib32z-dev libgl1-mesa-dev xsltproc unzip \
  mosh htop lsof screen -y

git clone --depth=1 https://github.com/MiCode/Xiaomi_Kernel_OpenSource.git -b $BRANCH $BRANCH

cd $BRANCH
mkdir out

git clone https://android.googlesource.com/platform/external/dtc -b pie-release
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9 toolchain
tar xvf snapdragon-llvm-6.0.9-linux64.tar.gz

# DTC
git clone https://git.kernel.org/pub/scm/utils/dtc/dtc.git dtc-1.4.6 -b v1.4.6
cd dtc-1.4.6
make NO_PYTHON=1

export ARCH=arm64
export SUBARCH=arm64
export DTC_EXT=/root/raphael-p-oss/dtc-1.4.6/dtc
export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-android-
export CONFIG_BUILD_ARM64_DT_OVERLAY=y

make O=out REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- raphael_user_defconfig
make -j$(nproc) O=out DTC_EXT=dtc REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log
```

Here is the log
[kernel.log](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/3426783/kernel.log)

okitavera, 2019-07-24:
use `make xconfig` to set `CONFIG_BUILD_ARM64_DT_OVERLAY` to `y`

csrutil, 2019-07-24:
Thank you for your help and time. I am not good at it, so I want to try compile it. People always good at something, but not everything.

closed 2019-07-24 by csrutil