MiCode/Xiaomi_Kernel_OpenSource #1115: dtc编译错误

shandongtlb, 2019-07-24:
如题，dtc编译出现错误，版本是aosp的pie-release分支  
git clone https://android.googlesource.com/platform/external/dtc -b pie-release
报错如下
         DEP libfdt/fdt_wip.c
	 DEP libfdt/fdt_ro.c
	 DEP libfdt/fdt.c
	 DEP util.c
	 DEP fdtoverlay.c
	 DEP fdtput.c
	 DEP fdtget.c
	 DEP fdtdump.c
	 LEX convert-dtsv0-lexer.lex.c
	 DEP convert-dtsv0-lexer.lex.c
	 DEP srcpos.c
	 BISON dtc-parser.tab.c
dtc-parser.y: warning: 3 shift/reduce conflicts [-Wconflicts-sr]
	 DEP dtc-parser.tab.c
	 LEX dtc-lexer.lex.c
	 DEP dtc-lexer.lex.c
	 DEP treesource.c
	 DEP livetree.c
	 DEP fstree.c
	 DEP flattree.c
	 DEP dtc.c
	 DEP data.c
	 DEP checks.c
	 CC srcpos.o
	 CC util.o
	 CC convert-dtsv0-lexer.lex.o
	 LD convert-dtsv0
	 CC dtc.o
	 CC checks.o
	 CC data.o
	 CC flattree.o
	 CC fstree.o
	 CC livetree.o
	 CC treesource.o
make: *** No rule to make target 'dtc-parser.h', needed by 'dtc-lexer.lex.o'.  Stop.
求大佬帮忙

sekaiacg, 2019-07-25:
dtc-lexer.l中使用的头文件不正确：
#include "dtc-parser.h -> #include "dtc-parser.tab.h"

shandongtlb, 2019-07-25:
> dtc-lexer.l中使用的头文件不正确：
> #include "dtc-parser.h -> #include "dtc-parser.tab.h"

哇谢谢大佬

shandongtlb, 2019-07-25:
> dtc-lexer.l中使用的头文件不正确：
> #include "dtc-parser.h -> #include "dtc-parser.tab.h"

已经可以编译出Image.gz-dtb和ko模块了，但到最后蹦出一大堆关于dtb警告是怎么肥四😅？
dtc用的aosp的pie-release

sekaiacg, 2019-08-02:
可以屏蔽掉警告，没什么影响
./dtc --help

shandongtlb, 2019-08-04:
> 可以屏蔽掉警告，没什么影响
> ./dtc --help

好的谢谢大佬

closed 2019-08-04 by shandongtlb

krafi99, 2021-04-06:
> dtc-lexer.l中使用的头文件不正确：
> #include "dtc-parser.h -> #include "dtc-parser.tab.h"

I'm also facing same issue. However, after doing this it shows another error:

```
$ make
         DEP tests/dumptrees.c
         DEP tests/trees.S
         DEP tests/testutils.c
         DEP tests/value-labels.c
         DEP tests/asm_tree_dump.c
         DEP tests/truncated_property.c
         DEP tests/check_path.c
         DEP tests/overlay_bad_fixup.c
         DEP tests/overlay.c
         DEP tests/subnode_iterate.c
         DEP tests/property_iterate.c
         DEP tests/integer-expressions.c
         DEP tests/utilfdt_test.c
         DEP tests/path_offset_aliases.c
         DEP tests/add_subnode_with_nops.c
         DEP tests/dtbs_equal_unordered.c
         DEP tests/dtb_reverse.c
         DEP tests/dtbs_equal_ordered.c
         DEP tests/extra-terminating-null.c
         DEP tests/incbin.c
         DEP tests/boot-cpuid.c
         DEP tests/phandle_format.c
         DEP tests/path-references.c
         DEP tests/references.c
         DEP tests/string_escapes.c
         DEP tests/propname_escapes.c
         DEP tests/appendprop2.c
         DEP tests/appendprop1.c
         DEP tests/del_node.c
         DEP tests/del_property.c
         DEP tests/setprop.c
         DEP tests/set_name.c
         DEP tests/rw_tree1.c
         DEP tests/open_pack.c
         DEP tests/nopulate.c
         DEP tests/mangle-layout.c
         DEP tests/move_and_save.c
         DEP tests/sw_tree1.c
         DEP tests/nop_node.c
         DEP tests/nop_property.c
         DEP tests/setprop_inplace.c
         DEP tests/stringlist.c
         DEP tests/addr_size_cells.c
         DEP tests/notfound.c
         DEP tests/sized_cells.c
         DEP tests/char_literal.c
         DEP tests/get_alias.c
         DEP tests/node_offset_by_compatible.c
         DEP tests/node_check_compatible.c
         DEP tests/node_offset_by_phandle.c
         DEP tests/node_offset_by_prop_value.c
         DEP tests/parent_offset.c
         DEP tests/supernode_atdepth_offset.c
         DEP tests/get_path.c
         DEP tests/get_phandle.c
         DEP tests/getprop.c
         DEP tests/get_name.c
         DEP tests/path_offset.c
         DEP tests/subnode_offset.c
         DEP tests/find_property.c
         DEP tests/root_node.c
         DEP tests/get_mem_rsv.c
         DEP libfdt/acpi.c
         DEP libfdt/fdt_overlay.c
         DEP libfdt/fdt_addresses.c
         DEP libfdt/fdt_empty_tree.c
         DEP libfdt/fdt_strerror.c
         DEP libfdt/fdt_rw.c
         DEP libfdt/fdt_sw.c
         DEP libfdt/fdt_wip.c
         DEP libfdt/fdt_ro.c
         DEP libfdt/fdt.c
         DEP util.c
         DEP fdtoverlay.c
         DEP fdtput.c
         DEP fdtget.c
         DEP fdtdump.c
         LEX convert-dtsv0-lexer.lex.c
         DEP convert-dtsv0-lexer.lex.c
         DEP srcpos.c
         BISON dtc-parser.tab.c
dtc-parser.y: warning: 3 shift/reduce conflicts [-Wconflicts-sr]
dtc-parser.y: note: rerun with option '-Wcounterexamples' to generate conflict counterexamples
         DEP dtc-parser.tab.c
         LEX dtc-lexer.lex.c
         DEP dtc-lexer.lex.c
         DEP treesource.c
         DEP livetree.c
         DEP fstree.c
         DEP flattree.c
         DEP dtc.c
         DEP data.c
         DEP checks.c
         CC srcpos.o
         CC util.o
         CC convert-dtsv0-lexer.lex.o
         LD convert-dtsv0
         CC dtc.o
         CC checks.o
         CC data.o
         CC flattree.o
         CC fstree.o
         CC livetree.o
         CC treesource.o
         CC dtc-lexer.lex.o
         CC dtc-parser.tab.o
         LD dtc
         CC fdtdump.o
         LD fdtdump
         CC fdtget.o
         CC libfdt/fdt.o
         CC libfdt/fdt_ro.o
         CC libfdt/fdt_wip.o
         CC libfdt/fdt_sw.o
         CC libfdt/fdt_rw.o
         CC libfdt/fdt_strerror.o
         CC libfdt/fdt_empty_tree.o
         CC libfdt/fdt_addresses.o
         CC libfdt/fdt_overlay.o
         CC libfdt/acpi.o
libfdt/acpi.c: In function ‘acpi_csum’:
libfdt/acpi.c:38:6: error: cast discards ‘const’ qualifier from pointer target type [-Werror=cast-qual]
   38 |  p = (uint8_t*)base;
      |      ^
cc1: all warnings being treated as errors
make: *** [Makefile:316: libfdt/acpi.o] Error 1
```