MiCode/Xiaomi_Kernel_OpenSource #1116: Mi 9T / K20 kernel and DTS are broken

MirkoCovizzi, 2019-07-25:
Hi everyone!
I've successfully compiled the kernel, however I noticed that the samsung display device tree source is linked to nothing in the K20 source. This results in a compiled (through the official DTB tool mkdtboimg) dtbo.img that doesn't boot properly, lacking all the timings and informations of the samsung display.
There seems to be a profound issue in the kernel source, especially since there is a substantial difference in the commit concerning the Samsung display between the K20 Pro and the K20.

MirkoCovizzi, 2019-07-25:
Made a breakthrough:
Unpacked dtbo.img from global firmware 10.3.6. Decoded the .dtb into .dts. Taken the 13 .dts relevant to the device (check Makefile). Re-encoded and repacked in a new, smaller dtbo.img.
I flashed this new dtbo.img through fastboot, booting successfully the device.
By taking the .dts provided by Xiaomi in the Mi 9T / K20 repo and compiling them, then packing them into a dtbo.img, and then flashing said .img through fastboot causes an unsuccessfull boot and a black screen.
So, I took the first .dts (from the source) and compared it to the correspondent .dts  (from the decoded global firmware) and got this interesting diff:

2973,2974c2973,2974
< 			qcom,mdss_dsi_ss_fhd_ea_f10_cmd {
< 				qcom,mdss-dsi-panel-name = "ss ea fhd cmd dsi panel";
---
> 			qcom,mdss_dsi_ss_fhd_ea8076_f10_cmd {
> 				qcom,mdss-dsi-panel-name = "samsung ea8076 fhd cmd dsi panel";
2976c2976
< 				qcom,mdss-dsi-panel-model = "SS FHD EA CMD PANEL";
---
> 				qcom,mdss-dsi-panel-model = "SAMSUNG FHD EA8076 CMD PANEL";
2997c2997
< 				qcom,mdss-dsi-bl-dcs-type-ss-ea;
---
> 				qcom,mdss-dsi-bl-dcs-type-ss;
3107,3108c3107,3108
< 			qcom,mdss_dsi_ss_fhd_eb_f10_cmd {
< 				qcom,mdss-dsi-panel-name = "ss eb fhd cmd dsi panel";
---
> 			qcom,mdss_dsi_ss_fhd_r66456_f10_cmd {
> 				qcom,mdss-dsi-panel-name = "ss r66456 fhd cmd dsi panel";
3110c3110
< 				qcom,mdss-dsi-panel-model = "SS FHD EB CMD PANEL";
---
> 				qcom,mdss-dsi-panel-model = "SS FHD R66456 CMD PANEL";
3132c3132
< 				qcom,mdss-dsi-bl-dcs-type-ss-eb;
---
> 				qcom,mdss-dsi-bl-dcs-type-r66456;
3148,3149d3147
< 				qcom,local-hbm-enabled;
< 				qcom,mdss-panel-on-dimming-delay = <0x78>;
3183c3181
< 						qcom,mdss-dsi-on-command = [39 01 00 00 00 00 02 b0 00 39 00 00 00 00 00 2c bb 59 ff ff ff ef cf ab 87 63 3f 4a 48 46 44 42 40 3e 3c 3a 64 00 ff ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 02 02 42 00 39 00 00 00 00 00 03 d4 00 8d 39 00 00 00 00 00 02 fa 0f 39 01 00 00 00 00 02 b0 80 29 00 00 00 00 00 0b cc 10 00 00 00 00 00 00 fe 03 00 29 01 00 00 00 00 87 e2 2b 00 ff ff ff 00 ff ff ff 00 ff 00 00 00 ff 80 ff 00 ff ff ff ff 00 5f 00 5f 02 ff 00 5a a0 00 00 33 00 04 00 1b 00 1e 00 1b a0 1e 00 00 00 3c 00 00 00 5f 00 00 a0 5f 00 00 00 83 00 1c 00 a2 00 1c a0 a2 00 37 00 bd 00 5e 00 be 10 60 f0 00 00 89 00 02 00 a2 00 1c 10 a2 f0 1c 00 bd 00 39 00 bd 00 5f 10 bd f0 5f 00 bd 00 88 00 a2 00 a2 10 a2 f0 a2 00 84 00 be 00 63 00 be 03 02 1c 07 ea 00 00 00 00 00 39 00 00 00 00 00 02 e6 00 39 00 00 00 00 00 02 35 00 39 00 00 00 00 00 05 2a 00 00 04 37 39 00 00 00 00 00 05 2b 00 00 09 23 39 00 00 00 00 00 03 51 01 00 05 01 00 00 6e 00 02 11 00 05 01 00 00 00 00 02 29 00];
---
> 						qcom,mdss-dsi-on-command = [39 01 00 00 00 00 02 b0 00 39 00 00 00 00 00 2c bb 59 ff ff ff ef cf ab 87 63 3f 4a 48 46 44 42 40 3e 3c 3a 64 00 ff ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 00 02 02 42 00 39 00 00 00 00 00 03 d4 00 8d 39 00 00 00 00 00 02 fa 0f 39 01 00 00 00 00 02 b0 80 39 00 00 00 00 00 02 e6 00 39 00 00 00 00 00 02 35 00 39 00 00 00 00 00 05 2a 00 00 04 37 39 00 00 00 00 00 05 2b 00 00 09 23 39 00 00 00 00 00 03 51 01 00 05 01 00 00 6e 00 02 11 00 05 01 00 00 10 00 02 29 00 39 01 00 00 00 00 02 b0 00 39 01 00 00 00 00 04 f0 cf c3 00 39 01 00 00 00 00 04 f0 cf c4 00];
3187c3185
< 						qcom,mdss-dsi-dispparam-hbm-fod-off-command = [29 01 00 00 00 00 02 b0 80 29 01 00 00 00 00 02 cc 10];
---
> 						qcom,mdss-dsi-dispparam-hbm-fod-off-command = [39 01 00 00 00 00 02 53 20];
3189c3187
< 						qcom,mdss-dsi-dispparam-hbm-fod-on-command = [39 00 00 00 00 00 02 88 00 29 01 00 00 00 00 02 b0 80 29 00 00 00 00 00 0b e1 01 00 21 1a 0f ff 00 00 00 02 29 01 00 00 00 00 02 cc 13];
---
> 						qcom,mdss-dsi-dispparam-hbm-fod-on-command = [39 00 00 00 00 00 02 88 00 39 01 00 00 00 00 02 53 e0];
3191,3195c3189,3191
< 						qcom,mdss-dsi-dispparam-lbm-fod-on-command = [39 00 00 00 00 00 02 88 00 29 01 00 00 00 00 02 b0 80 29 00 00 00 00 00 0b e1 01 00 21 18 0b ff 00 00 00 02 29 01 00 00 00 00 02 cc 13];
< 						qcom,mdss-dsi-dispparam-lbm-fod-on-command-state = "dsi_hs_mode";
< 						qcom,mdss-dsi-doze-hbm-command = [39 00 00 00 00 00 03 51 00 ff 39 01 00 00 00 00 02 88 01];
< 						qcom,mdss-dsi-doze-lbm-command = [39 00 00 00 00 00 03 51 00 00 39 01 00 00 00 00 02 88 01];
< 						qcom,mdss-dsi-nolp-command = [39 01 00 00 00 00 02 88 00];
---
> 						qcom,mdss-dsi-doze-hbm-command = [39 01 00 00 00 00 03 51 00 ff 39 01 00 00 00 00 02 b0 00 39 01 00 00 00 00 04 f0 bb 2a 40 39 01 00 00 00 00 02 53 20];
> 						qcom,mdss-dsi-doze-lbm-command = [39 01 00 00 00 00 03 51 00 10 39 01 00 00 00 00 02 b0 00 39 01 00 00 00 00 04 f0 bb 2a 40 39 01 00 00 00 00 02 53 20];
> 						qcom,mdss-dsi-nolp-command = <0x39010000 0x351 0x1003901 0x00 0x2b00039 0x1000000 0x4f0bb 0x2a403901 0x100 0x2532039 0x1000000 0x28800>;
3198a3195,3196
> 						qcom,mdss-dsi-lp1-command = [39 01 00 00 00 00 03 51 00 00 39 01 00 00 00 00 02 88 01];
> 						qcom,mdss-dsi-lp1-command-state = "dsi_lp_mode";
3207c3205
< 						qcom,mdss-dsi-dispparam-hbm-off-command = [39 01 00 00 00 00 02 53 28];
---
> 						qcom,mdss-dsi-dispparam-hbm-off-command = [39 01 00 00 ff 00 02 53 28 39 01 00 00 00 00 02 53 20];
3211,3214d3208
< 						qcom,mdss-dsi-dispparam-dimmingon-command = [39 01 00 00 01 00 02 53 28];
< 						qcom,mdss-dsi-dispparam-dimmingon-command-state = "dsi_lp_mode";
< 						qcom,mdss-dsi-dispparam-dimmingoff-command = [39 01 00 00 01 00 02 53 20];
[... 22 lines cut ...]
[... cut ...] 0/fragment@44:target:0\0/fragment@45/__overlay__/gpio_keys/vol_up:gpios:0\0/fragment@50/__overlay__/gpio-regulator@1:gpio:0\0/fragment@51/__overlay__/qcom,cam-sensor@0:gpios:24\0/fragment@51/__overlay__/qcom,cam-sensor@1:gpios:24\0/fragment@68:target:0";
5582c5577
< 				qcom,mdss_dsi_ss_fhd_ea_f10_cmd {
---
> 				qcom,mdss_dsi_ss_fhd_ea8076_f10_cmd {
5592c5587
< 				qcom,mdss_dsi_ss_fhd_eb_f10_cmd {
---
> 				qcom,mdss_dsi_ss_fhd_r66456_f10_cmd {

So, there is obviously something wrong going on.

Zenderable, 2019-07-27:
I confirm that there is a problem.

MirkoCovizzi, 2019-07-28:
Update:

The device tree clearly has some problems. However, the kernel is clearly broken too. Flashed over eu stable 10.3.12.0 without changing dtbo doesn't boot. Flashed over eu stable 10.3.12.0 together with the dtbo image obtained compiling the source device tree doesn't boot too (bootloop).

PIPIPIG233666, 2019-07-28:
@ivan19871002

PIPIPIG233666, 2019-07-28:
> Made a breakthrough:
> Unpacked dtbo.img from global firmware 10.3.6. Decoded the .dtb into .dts. Taken the 13 .dts relevant to the device (check Makefile). Re-encoded and repacked in a new, smaller dtbo.img.
> I flashed this new dtbo.img through fastboot, booting successfully the device.
> By taking the .dts provided by Xiaomi in the Mi 9T / K20 repo and compiling them, then packing them into a dtbo.img, and then flashing said .img through fastboot causes an unsuccessfull boot and a black screen.
> So, I took the first .dts (from the source) and compared it to the correspondent .dts (from the decoded global firmware) and got this interesting diff (green one is the one working, red one is from the source that doesn't work):
> 
> https://pastebin.com/wNu0mVy0
> 
> So, there is obviously something wrong going on, concerning the display commands.

It seems there are two different dts ,one is samsung ,another is ss.
[https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/davinci-p-oss/arch/arm64/boot/dts/qcom/dsi-panel-samsung-fhd-ea8076-f3b-cmd.dtsi](url)
[https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/davinci-p-oss/arch/arm64/boot/dts/qcom/dsi-panel-ss-fhd-ea-f10-cmd.dtsi](url)

MirkoCovizzi, 2019-07-28:
> It seems there are two different dts ,one is samsung ,another is ss. [...]

It's the second one the one to consider (which is still a Samsung display, by the way). It is the one that gets compiled in the dtbo.

MirkoCovizzi, 2019-07-30:
@PIPIPIG233666 I'm not completely sure adding the f3b-cmd.dtsi would solve the issues. In the first diff, some commands used by the stock kernel (based on miui.eu 10.3.12.0) are even different from the ones found in the f3b-cmd.dtsi.

MirkoCovizzi, 2019-08-02:
Update: on the lineageos port I did a couple of days ago both kernel and dts work. No change needed. It appears it can't work on miui only for some reason.

closed 2019-08-02 by MirkoCovizzi