MiCode/Xiaomi_Kernel_OpenSource #1125: Perseus branch builds kernel that does not boot

xpwn3rx, 2019-08-15:
I am compiling the base perseus kernel from the branch here, which I have forked. The kernel builds (after removing the warning stopping code in gcc-wrapper.py) but the kernel does not boot. I am wanting to modify the stock MIUI based kernel.

I built using these command lines:

ake O=out clean && make O=out mrproper && make O=out perseus_user_defconfig && make O=out -j16 DTC_EXT=dtc CONFIG_BUILD_ARM64_DT_OVERLAY=y

liuwang97, 2019-08-16:
The kernel source is out of date.    
in the lastset MIUI，lmkd  need psi monitor to start up.  otherwise, devices will boot into fastboot.   please  add this patch for your kernel.   https://lwn.net/Articles/774969/
or just set the persist.lmk.use_psi property  to  false.

xpwn3rx, 2019-08-17:
Thank you, I will try that.