MiCode/Xiaomi_Kernel_OpenSource #1127: dipper米8内核编译问题

qq995473428, 2019-08-24:
编译完成后使用eto内核包替换里面编译好的dtb文件刷入后无限重启怎么破

qq995473428, 2019-08-24:
求大佬

YumeMichi, 2019-08-24:
内核源码已经不适用于开发版9.7.24之后的系统了

qq995473428, 2019-08-24:
> 内核源码已经不适用于开发版9.7.24之后的系统了

那要咋办哦，我的是9.7.13

YumeMichi, 2019-08-24:
> > 内核源码已经不适用于开发版9.7.24之后的系统了
> 
> 那要咋办哦，我的是9.7.13

用系统自带的 或者别的第三方内核吧

qq995473428, 2019-08-24:
> > > 内核源码已经不适用于开发版9.7.24之后的系统了
> > 
> > 
> > 那要咋办哦，我的是9.7.13
> 
> 用系统自带的 或者别的第三方内核吧

- - 不对呀酷安发布的内核是咋编译的刷最新的系统都可以用啊

YumeMichi, 2019-08-24:
> > > > 内核源码已经不适用于开发版9.7.24之后的系统了
> > > 
> > > 
> > > 那要咋办哦，我的是9.7.13
> > 
> > 
> > 用系统自带的 或者别的第三方内核吧
> 
> * * 不对呀酷安发布的内核是咋编译的刷最新的系统都可以用啊

那不就是第三方内核了 又不是直接用这个代码编译的

qq995473428, 2019-08-24:
> > > > > 内核源码已经不适用于开发版9.7.24之后的系统了
> > > > 
> > > > 
> > > > 那要咋办哦，我的是9.7.13
> > > 
> > > 
> > > 用系统自带的 或者别的第三方内核吧
> > 
> > 
> > 
> > * * 不对呀酷安发布的内核是咋编译的刷最新的系统都可以用啊
> 
> 那不就是第三方内核了 又不是直接用这个代码编译的

那是用啥内核编译的

YumeMichi, 2019-08-24:
> > > > > > 内核源码已经不适用于开发版9.7.24之后的系统了
> > > > > 
> > > > > 
> > > > > 那要咋办哦，我的是9.7.13
> > > > 
> > > > 
> > > > 用系统自带的 或者别的第三方内核吧
> > > 
> > > 
> > > 
> > > * * 不对呀酷安发布的内核是咋编译的刷最新的系统都可以用啊
> > 
> > 
> > 那不就是第三方内核了 又不是直接用这个代码编译的
> 
> 那是用啥内核编译的

自己修改的内核=。=

qq995473428, 2019-08-24:
> > > > > > > 内核源码已经不适用于开发版9.7.24之后的系统了
> > > > > > 
> > > > > > 
> > > > > > 那要咋办哦，我的是9.7.13
> > > > > 
> > > > > 
> > > > > 用系统自带的 或者别的第三方内核吧
> > > > 
> > > > 
> > > > 
> > > > * * 不对呀酷安发布的内核是咋编译的刷最新的系统都可以用啊
> > > 
> > > 
> > > 那不就是第三方内核了 又不是直接用这个代码编译的
> > 
> > 
> > 那是用啥内核编译的
> 
> 自己修改的内核=。=

看很多教程都是官方给的源码编译的

Gah0, 2019-10-17:
内核过时已经不适配最新的MIUI版本。
只有正确的Device Tree binary才能使Bootloader引导内核启动

Gah0, 2019-10-17:
> 内核过时已经不适配最新的MIUI版本。
> 只有正确的设备树二进制才能使Bootloader的引导内核启动

另外，关闭lmkd可能也能解决问题。