MiCode/Xiaomi_Kernel_OpenSource #1129: 米9编译问题

shandongtlb, 2019-09-05:
求救大佬
在我勾选 network=>wireless=>IEEE80211cfg这个选项之后就报错了
  AR      net/wireless/cfg80211.o
  AR      net/wireless/built-in.o
  CC      drivers/hid/hid-trace.o
  CC      drivers/gpu/msm/kgsl_mmu.o
  CC      drivers/gpu/msm/kgsl_snapshot.o
  CC      drivers/gpu/drm/drm_scdc_helper.o
  AR      drivers/hid/hid.o
  CC      drivers/gpu/drm/drm_gem_framebuffer_helper.o
../net/mac80211/tx.c:814:20: error: no member named 'mntr_flags' in 'union ieee80211_sub_if_data::(anonymous at ../net/mac80211/ieee80211_i.h:952:2)'
           !(tx->sdata->u.mntr_flags & MONITOR_FLAG_COOK_FRAMES))) {
             ~~~~~~~~~~~~ ^
../include/linux/compiler.h:77:42: note: expanded from macro 'unlikely'
# define unlikely(x)    __builtin_expect(!!(x), 0)
                                            ^
1 error generated.
make[3]: *** [../scripts/Makefile.build:363: net/mac80211/tx.o] Error 1
make[2]: *** [../scripts/Makefile.build:678: net/mac80211] Error 2
make[2]: *** Waiting for unfinished jobs....

不知道什么问题，求救大佬

shandongtlb, 2019-09-05:
好像是缺什么函数还是什么值不对，求救大佬们

closed 2019-09-06 by shandongtlb

OpPRO-official, 2022-02-19:
How you fix this error?