MiCode/Xiaomi_Kernel_OpenSource #1133: pine-p-oss branch succesfully builds kernel that boots straight to fastboot

YaAlex3, 2019-09-28:
Hello. I'm trying to build Redmi 7A kernel standalone, getting kernel that doesn't boots.
My environment and compilation script:
```
DTC_EXT="/home/yaalex3/android/dtc/dtc"
CROSS_COMPILE="/home/yaalex3/android/gcc4.932/arm-linux-androideabi-4.9-refs_heads_pie-release/bin/arm-linux-androideabi-"
ARCH=arm
SUBARCH=arm
CONFIG_BUILD_ARM64_DT_OVERLAY="y"
```
```
make O=out pine-perf_defconfig
make O=out -j16 -l16
```
The question is, am i doing something wrong or kernel source has issues?

YaAlex3, 2019-09-28:
UPD: I managed to boot by getting dtbs from stock kernel using extract_dtb tool from aosp prebuilts/misc/libufdt, then appended these to my compiled zImage
anyone knows what could be wrong there?

ivan19871002, 2019-11-01:
CONFIG_BUILD_ARM64_DT_OVERLAY=y  add  into config

closed 2019-11-01 by ivan19871002

GuaiYiHu, 2019-11-11:
https://github.com/GuaiYiHu/android_kernel_xiaomi_sdm439/commit/84474583368d23d4dce507d237edfc2dda0b2370
https://github.com/GuaiYiHu/android_kernel_xiaomi_sdm439/commit/8b596c760e39f28a1240195e24aeb611dcf637b4