MiCode/Xiaomi_Kernel_OpenSource #1139: cepheus-p-oss 编译问题

shenzhenkevin, 2019-10-09:
环境：Ubuntu 14.04 
1、下载源码zip包（github上下载），解压后使用NDK R17C包（官网下载）中的交叉编译工具进行编译；
工具路径：toolchains/aarch64-linux-android-4.9/prebuilt/linux-x86_64/bin
2、报错如下：
make[1]: Entering directory `/root/Xiaomi_Kernel_OpenSource-cepheus-p-oss/out'
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  Using .. as source for kernel
  GEN     ./Makefile
  ....
  CC      [kernel/bounds.s]
Android GCC has been deprecated in favor of Clang, and will be removed from
Android in 2020-01 as per the deprecation plan in:
https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/master/GCC_4_9_DEPRECATION.md

921d9846-a815-11e9-84ea-ff8bce1b90b6: error: Android: No such file or directory
921d9846-a815-11e9-84ea-ff8bce1b90b6: error: GCC: No such file or directory
921d9846-a815-11e9-84ea-ff8bce1b90b6: error: has: No such file or directory
921d9846-a815-11e9-84ea-ff8bce1b90b6: error: been: No such file or directory
......
make[2]: *** [kernel/bounds.s] Error 1
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory `/root/Xiaomi_Kernel_OpenSource-cepheus-p-oss/out'
make: *** [sub-make] Error 2
make: Leaving directory `/root/Xiaomi_Kernel_OpenSource-cepheus-p-oss'
[错误提示.txt](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/3705606/default.txt)
3、编译执行的脚本：
#!/bin/bash

OUT_DIR=out
COMMON_ARGS="-C $(pwd) O=$(pwd)/${OUT_DIR} ARCH=arm64 CROSS_COMPILE=aarch64-linux-android- KCFLAGS=-mno-android"

#通过git下载 https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9
export PATH=$PATH:/root/git_gcc_linun_x86/bin

[ -d ${OUT_DIR} ] && rm -rf ${OUT_DIR}
mkdir ${OUT_DIR}

#make ${COMMON_ARGS} star2qlte_chn_open_defconfig
make ${COMMON_ARGS} cepheus_user_defconfig
make ${COMMON_ARGS}

#cp ${OUT_DIR}/arch/arm64/boot/Image $(pwd)/arch/arm64/boot/Image

shenzhenkevin, 2019-10-26:
参照wiki 独立编译指导

closed 2019-10-26 by shenzhenkevin

daddas, 2022-03-09:
我也遇到过，建议修改下编译目录中的MakeFile 去掉 gccwrapper部分，然后修改下编译flag

具体：

`REAL_CC		= $(CROSS_COMPILE)gcc` 改为 `CC		= $(CROSS_COMPILE)gcc`

去掉
`CC		= $(srctree)/scripts/gcc-wrapper.py $(REAL_CC)`

改`KBUILD_CFLAGS   := -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs \`为`KBUILD_CFLAGS   := -Werror -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs \`。