MiCode/Xiaomi_Kernel_OpenSource #1147: cepheus编译过程dtc问题

shenzhenkevin, 2019-10-24:
参照wiki指导编译。第一次出现的错误是“dtc: command not found”，通过apt-get install安装了1.4.0版本的dtc。再编译后上述错误消失，但出现新错误：“dtc: invalid option -- '@'”。
   怀疑是否安装的dtc版本太老？但查询网上信息，说是dtc是kernel scripts下自带的，我查询scripts/dtc/version_gen.h中显示“ DTC 1.4.4-g756ffc4f”。这是说需要使用1.4.4版本的DTC？

shenzhenkevin, 2019-10-24:
将dtc [路径从新设置为AOSP的路径（Qualcomm_Android/prebuilts/misc/linux-x86/dtc），编译结果失败]。怀疑是dtc版本有问题。我查看dtc版本是Version: DTC 1.4.2-Android-build。另外查看kernel代码scripts/dtc目录显示如下：
cepheus-p-oss#more scripts/dtc/version_gen.h 
#define DTC_VERSION "DTC 1.4.4-g756ffc4f"

失败信息：
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
net/packet/af_packet.o: In function `check_copy_size':
/usr1/new-wiki/cepheus-p-oss/out/../include/linux/thread_info.h:136: undefined reference to `__bad_copy_to'
make[1]: *** [vmlinux] Error 1
make[1]: Leaving directory `/usr1/new-wiki/cepheus-p-oss/out'
make: *** [sub-make] Error 2

PIPIPIG233666, 2019-10-25:
兄弟你能不能把环境装全

PIPIPIG233666, 2019-10-25:
给你个装环境的[脚本](https://github.com/akhilnarang/scripts)

LenHackintosh, 2019-10-25:
> 兄弟你能不能把环境装全

兄弟请问member techpack/audio/dsp/elliptic in archive is not an object这种报错咋搞。编译包含audio/dsp/elliptic最后快编译完成的时候就出这个错误。

LenHackintosh, 2019-10-25:
dtc版本太老了。最新1.5.0

shenzhenkevin, 2019-10-26:
使用AOSP自带DTC（建一个AOSP工程，执行source build/envsetup.sh）。说是独立编译，但千万别信这鬼话。

closed 2019-10-26 by shenzhenkevin