MiCode/Xiaomi_Kernel_OpenSource #1151: Source code kernel  mi 9 android q

EviraKernel, 2019-11-01:
pls when is relase it source code for kernel android q mi 9

thune-xiaobai, 2019-11-01:
> pls when is relase it source code for kernel android q mi 9

After MIUI q stable version released.

mrslezak, 2019-11-06:
So MIUI 11 Stable is fully released.  For Cepheus and Raphael.  Where is the Android 10 kernel source?  GPL requirement to post source when used by a company in their software.  What's the timeline?

abdoi94, 2019-11-07:
> > pls when is relase it source code for kernel android q mi 9
> 
> After MIUI q stable version released.

It's already released in multiple regions 🤦‍♂️

Also beta/weekly based on Q has been going for +3 months now, and no signs of kernel source, GPL should apply to beta releases also

mrslezak, 2019-11-07:
Well we have the GPL enforcement lawyers in Germany on standby just wanting an answer from Xiaomi on the release date.

abdoi94, 2019-11-12:
👏where👏is👏the👏 kernel👏source👏
👏 global👏 ROMs👏 already👏got👏 released👏

RobMacGregor, 2019-11-16:
Looking forward to the kernel as well.

ivan19871002, 2019-11-20:
this month .

brituj, 2019-11-26:
No more delays. Give kernel source ASAP!!

PIPIPIG233666, 2019-11-27:
> No more delays. Give kernel source ASAP!!

this week afaik

LenHackintosh, 2019-11-28:
> this month .

This month is almost over.....Please keep your promise

Kugumin, 2019-11-29:
The time will come soon, we are waiting for sources

LenHackintosh, 2019-11-30:
the last day..So it's really a monkey-playing company

1582130940, 2019-11-30:
Expected to be released on Monday, December 1st

AndroMor, 2019-11-30:
This is my last Xiaomi phone. Why no kernel till now  Really slow Xiaomi 🤬

closed 2019-12-01 by ivan19871002