MiCode/Xiaomi_Kernel_OpenSource #1155: K20 Pro/Mi 9T Pro Q kernel sources

Kugumin, 2019-11-21:
Since GS (MI) update channel got Android 10 all devs are expecting to see updated sources in near future!

abdoi94, 2019-11-21:
this month .

_Originally posted by @ivan19871002 in https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/1151#issuecomment-555961837_

dcionline, 2019-11-21:
let them patching first issues, or want buggy sources instead? Damn all has to be fast ans after its bad if it doesnt work.

abdoi94, 2019-11-21:
> let them patching first issues, or want buggy sources instead? Damn all has to be fast and after its bad if it doesnt work.

They have been releasing ten builds since early August with Miui weekly builds, I'm pretty sure they have done enough work on it already

addhaloka, 2019-11-22:
> let them patching first issues, or want buggy sources instead? Damn all has to be fast and after its bad if it doesnt work.
the opened sources will never get patched, you have to fix yourself

pappschlumpf, 2019-11-28:
I would be very pleased too if you would release the Mi9 MI9T source asap.

AndroMor, 2019-11-30:
I can't wait longer for Raphael kernel sources I am moving to another phone other than Xiaomi 🤬 slow

pappschlumpf, 2019-11-30:
Well Xiaomi, what's going on? Could be that difficult to upload sources?!

abdoi94, 2019-11-30:
Woah guys, wait for Monday to come at least

Kugumin, 2019-11-30:
It's already 1st December in China 😁

abdoi94, 2019-11-30:
> It's already 1st December in China

Don't expect anything on Sunday 🌚

UtsavisGreat, 2019-12-01:
> > It's already 1st December in China
> 
> Don't expect anything on Sunday

;_;

abdoi94, 2019-12-01:
> > It's already 1st December in China
> 
> Don't expect anything on Sunday

oh boi, that didnt age well

Kugumin, 2019-12-01:
Sources are released. Closing issue 😁

closed 2019-12-01 by Kugumin