MiCode/Xiaomi_Kernel_OpenSource #1156: [pine-p-oss] fastboot error when booting kernel built from source ("dtb not found")

jimvonmoon, 2019-11-24:
I am unable to boot Redmi 7A with kernel built from source. After trying device stays at bootloader screen and reports the following error:
```
> fastboot boot zImage-dtb
creating boot image...
creating boot image - 21803008 bytes
Sending 'boot.img' (21292 KB)                      OKAY [  0.671s]
Booting                                            FAILED (remote: 'dtb not found')
fastboot: error: Command failed
```

Kernel branch: `pine-p-oss`

Builder host system: `Ubuntu 19.10`

Toolchain used for build: `https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/+/refs/heads/pie-release`

Environment:
```
export CROSS_COMPILE=<path>/bin/arm-linux-androideabi-
export ARCH=arm
export SUBARCH=arm
export DTC_EXT=dtc
```

Build command:
```
make O=out pine-perf_defconfig
make O=out -j8
```

Other info:
- I already added `CONFIG_BUILD_ARM64_DT_OVERLAY=y` to `arch/arm/configs/pine-perf_defconfig` (as suggested in issue #1133).
- DTBs seem to be present at the end of `zImage-dtb`.
- I tried using dtc built from source, with no success (`https://android.googlesource.com/platform/external/dtc/+/refs/heads/pie-release`).
- I also tried making a simple `boot.img` by combining `zImage-dtb` with bare minimum initramfs and simple cmdline using `mkbootimg`. Same error.
- Bootloader seems to be unlocked:
```
> fastboot oem device-info
(bootloader)    Device tampered: false
(bootloader)    Device unlocked: true
(bootloader)    Device critical unlocked: true
(bootloader)    Charger screen enabled: false
(bootloader)    Display panel:
```

YaAlex3, 2019-11-24:
#1133

jimvonmoon, 2019-11-24:
Yes, I already mentioned #1133. Description and comments helped me a lot, but it doesn't really solve my problem.

ElDavoo, 2019-12-04:
Same thing with olive-p-oss. One thing I noticed is that dtb files built by us are not recognized by binwalk, while original dtbs are correctly recognized.

nikp123, 2020-07-20:
try setting "CONFIG_BUILD_ARM_APPENDED_DTB_IMAGE=y"

I can even ``fastboot boot`` the kernel now