MiCode/Xiaomi_Kernel_OpenSource #1159: Mi9 q source

EviraKernel, 2019-12-01:
I Compile latest mi9 q source no audio audio driver missing on source maybe need enable some?

bgcngm, 2019-12-01:
Dude, open your eyes: https://github.com/MiCode/vendor_qcom_opensource_audio-kernel

LenHackintosh, 2019-12-02:
need to import audio driver

EviraKernel, 2019-12-02:
i import audio driver but have problem with fp still not recognize wifi e audio no problem stock defconfig

LenHackintosh, 2019-12-03:
> i import audio driver but have problem with fp still not recognize wifi e audio no problem stock defconfig

force load modules

mrslezak, 2019-12-03:
He's saying the FP sensor isn't working after importing all the audio kernel and WiFi items.  Anyone else had this issue or a resolution?  Cepheus...

LenHackintosh, 2019-12-03:
> He's saying the FP sensor isn't working after importing all the audio kernel and WiFi items. Everything else is working now. Force loading modules. Anyone else had this issue or a resolution? Cepheus...

it's stable miui 11 kernel source,not dev，so if use it in the newest dev miui version,fp doesn't work.....in the stable MIUI11 or lower than 9.10.10 dev MIUI chaina version fp works. need xiaomi to solve it

dcionline, 2019-12-03:
take a log?

ivan19871002, 2019-12-04:
Now ,  Only support stable ,  not dev.

closed 2019-12-04 by ivan19871002