MiCode/Xiaomi_Kernel_OpenSource #1161: raphael-p-oss and cepheus-q-oss compile error

InPRTx, 2019-12-02:
root@localhost:~/git/Xiaomi_Kernel_OpenSource# clang -v
clang version 9.0.0-4ubuntu4
Target: aarch64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/aarch64-linux-gnu/8
Found candidate GCC installation: /usr/bin/../lib/gcc/aarch64-linux-gnu/9
Found candidate GCC installation: /usr/lib/gcc/aarch64-linux-gnu/8
Found candidate GCC installation: /usr/lib/gcc/aarch64-linux-gnu/9
Selected GCC installation: /usr/bin/../lib/gcc/aarch64-linux-gnu/9
Candidate multilib: .;@m64
Selected multilib: .;@m64

root@localhost:~/git/Xiaomi_Kernel_OpenSource# ld -V
GNU ld (GNU Binutils for Ubuntu) 2.33.1
  支持的仿真：
   aarch64linux
   aarch64elf
   aarch64elf32
   aarch64elf32b
   aarch64elfb
   armelf
   armelfb
   aarch64linuxb
   aarch64linux32
   aarch64linux32b
   armelfb_linux_eabi
   armelf_linux_eabi

apt install clang make m4 bison git
git clone https://github.com/MiCode/Xiaomi_Kernel_OpenSource --depth=1 -b cepheus-q-oss
cd Xiaomi_Kernel_OpenSource
make make O=out -j8 raphael_user_defconfig
make O=out -j8

Error log:
UPD     include/generated/compile.h
  CC      init/version.o
  AR      init/built-in.o
  AR      built-in.o
  LD      vmlinux.o
  MODPOST vmlinux.o
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
ld: drivers/platform/msm/gsi/gsi.o: relocation R_AARCH64_ABS32 against `__crc_gsi_write_channel_scratch' can not be used when making a shared object
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_complete_clk_grant+0x0): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_complete_clk_grant+0x8): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_map_base+0x0): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_map_base+0x8): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_unmap_base+0x0): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_unmap_base+0x8): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_register_device+0x0): 危 险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_register_device+0x8): 危 险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_write_device_scratch+0x0): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_write_device_scratch+0x8): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_deregister_device+0x0):  危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_deregister_device+0x8):  危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_alloc_evt_ring+0x0): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_alloc_evt_ring+0x8): 危险的重寻址: 不支持的重定位
drivers/platform/msm/gsi/gsi.o:(___ksymtab+gsi_write_evt_ring_scratch+0x0): 危险的重寻址: 不支持的重定位
...
make[1]: *** [/root/git/Xiaomi_Kernel_OpenSource/Makefile:1105：vmlinux] 错误 1
make[1]: 离开目录“/root/git/Xiaomi_Kernel_OpenSource/out”
make: *** [Makefile:146：sub-make] 错误 2

YumeMichi, 2019-12-02:
caf不支持clang吧

LenHackintosh, 2019-12-02:
use clang to build

InPRTx, 2019-12-02:
> use clang to build

Sorry,I use common is
make make O=out -j8 raphael_user_defconfig CC=clang
make O=out -j8 CC=clang
I forgot it the second time I edited it

LenHackintosh, 2019-12-02:
> > use clang to build
> 
> Sorry,I use common is
> make make O=out -j8 raphael_user_defconfig CC=clang
> make O=out -j8 CC=clang
> I forgot it the second time I edited it

aarch64-linux-gnu  use gcc7.4.1 or lower

sekaiacg, 2019-12-02:
编译gcc使用了 binutils 2.2.9 以上版本。
binutils 2.2.9以上会导致这个问题...emmmmmm

sekaiacg, 2019-12-02:
编译gcc使用了 binutils 2.29 以上版本。
binutils 2.29以上会导致这个问题...emmmmmm
还是希望xiaomi修复这个问题。

YumeMichi, 2019-12-02:
> 编译gcc使用了 binutils 2.29 以上版本。
> binutils 2.29以上会导致这个问题...emmmmmm
> 还是希望xiaomi修复这个问题。

跟小米没什么关系 qcom就没使用这么新的工具链 xiaomi也是

ivan19871002, 2019-12-04:
Not a issue. So close it 

pls use qcom's default toolchain

closed 2019-12-04 by ivan19871002

1498610052, 2020-04-23:
修复了吗，我也是这个问题.fst521.o也有问题Erro

hacker1024, 2020-07-27:
https://patchwork.kernel.org/patch/10708617/