MiCode/Xiaomi_Kernel_OpenSource #1162: No changeset for Poco F1 MIUI 10.3.8 (Android 9)

martins-pukitis, 2019-12-04:
Hi.
On my POCO F1 with original ROM:
`
beryllium:/ # uname -r
4.9.112-perf-g4aff1b2
`
It means I should compile the commit "4aff1b2" of the repo. But such commit doesn't exist in dipper-p-oss branch.

sekaiacg, 2019-12-06:
Xiaomi's open source code is different from the pre-compiled code in miui. So there's no corresponding commit

martins-pukitis, 2019-12-09:
Thanks for the reply. Is it possible to use Xiaomi's open source kernel with stock ROM? Are there any guides how to do it?

martins-pukitis, 2019-12-10:
I flashed the kernel compiled from latest changeset of dipper-p-oss branch (hash: 760a09a154b039a892a5ddd05906048f8ab295b9), but the device didn't boot. Switched back to original.