MiCode/Xiaomi_Kernel_OpenSource #1163: k20 Wifi does not work

rogan528, 2019-12-05:
12-02 19:03:05.462   721   721 E android.hardware.wifi@1.0-service: Wifi driver didn't get ready in time, giving up!: No such file or directory
12-02 19:03:05.462   721   721 E android.hardware.wifi@1.0-service: Failed to load WiFi driver
12-02 19:03:05.462   721   721 E android.hardware.wifi@1.0-service: Failed to initialize firmware mode controller
12-02 19:03:05.463   721   721 E android.hardware.wifi@1.0-service: Wifi HAL start failed

12-02 19:57:43.042   721   721 E android.hardware.wifi@1.0-service: Failed to write driver state control param: Invalid argument
12-02 19:57:43.042   721   721 E android.hardware.wifi@1.0-service: Failed to load WiFi driver
12-02 19:57:43.042   721   721 E android.hardware.wifi@1.0-service: Failed to initialize firmware mode controller
12-02 19:57:43.042   721   721 E android.hardware.wifi@1.0-service: Wifi HAL start failed
12-02 19:57:43.042  1229  1729 E HalDevMgr: Cannot start IWifi: 9 ()
12-02 19:57:43.042  1229  1706 E HalDevMgr: IWifiEventCallback.onFailure: 9 ()
12-02 19:57:43.042  1229  1729 E WifiVendorHal: Failed to start vendor HAL
12-02 19:57:43.042  1229  1729 E WifiNative: Failed to start vendor HAL
12-02 19:57:43.042  1229  1729 E WifiNative: Failed to start Hal

LenHackintosh, 2019-12-05:
force to load modules or you could merge qcacld-3.0

mrslezak, 2020-02-24:
CONFIG_FORCE_MODULES=y in defconfig as mentioned above + module.c is always busted for WiFi.  Look online for the commit to patch it.  Or qcacld works also as mentioned but that takes more work to implement.

yesimxev, 2020-03-07:
> [mrslezak/Cepheus-Raphael-Q-GCC10@521e8e9](https://github.com/mrslezak/Cepheus-Raphael-Q-GCC10/commit/521e8e9d393240ef76ae77c805305b40112e72c4) just follow what I did there it works for WiFi on Xiaomi devices.

Worked on a Nokia 3.1 too. Thank you very much!

mrslezak, 2020-03-07:
Anytime we have to help each other out or there's no progress :)

yesimxev, 2020-03-07:
Exactly! Looks like causes panic sometimes when I want to turn on wifi and reboots, needs some attention though but big progress in my project! Thanks again

yesimxev, 2020-03-16:
@mrslezak Looks like I need help to debug. When the phone turns on with wifi on, it's perfect, can turn on and off, connect etc.
When the phone starts with wifi off, it reboots when I press the wifi icon to turn on. After 3 reboots, it's working. But as soon as I want to turn off the wifi, it goes into a 10x reboot. Do you have any idea what can cause this?

mrslezak, 2020-03-16:
Wish I could help but on your Nokia phone, it's best to go to a Nokia support forum.  Just force WiFi to "on" boot and use that as a workaround in the meantime, or list it as a known bug in your kernel.

yesimxev, 2020-03-16:
Agree. My original moduke is fine though. I had to enable WEXT for wireless adapters, that's why module has to be force loaded. That's a module in vendor. So can't rebuild with WEXT enabled. I was thinking about the workaround, will try soon

LeCmnGend, 2022-01-19:
> [mrslezak/Cepheus-Raphael-Q-GCC10@521e8e9](https://github.com/mrslezak/Cepheus-Raphael-Q-GCC10/commit/521e8e9d393240ef76ae77c805305b40112e72c4) just follow what I did there it works for WiFi on Xiaomi devices.

I do it in a11 kernel and it doesnt work anymore, any solution for a11 sir ?

yesimxev, 2022-01-19:
Although the above forced the module to load, but still wasn't good enough for WEXT. I had to patch a line https://github.com/yesimxev/android_kernel_nokia_es2/commit/b6e78b3a8e4f41a1502a959691234a728901bef5 so external adapter's support wireless injection

LeCmnGend, 2022-01-19:
> Although the above forced the module to load, but still wasn't good enough for WEXT. I had to patch a line [yesimxev/android_kernel_nokia_es2@b6e78b3](https://github.com/yesimxev/android_kernel_nokia_es2/commit/b6e78b3a8e4f41a1502a959691234a728901bef5) so external adapter's support wireless injection while not interfering with internal wireless driver (so I don't need to force load and crash)

thank for your hint, let me try it <3