MiCode/Xiaomi_Kernel_OpenSource #1164: Could Xiaomi tell us what changed the fingerprint K20P and Mi 9 Q source in the dev (maybe future stable version)

LenHackintosh, 2019-12-05:
We know that the open source is based on the stable miui version, because the dev MIUI version is updated very quickly, and there are so many Xiaomi models. If the open source is based on the dev version, That would be a very onerous job。

But the current dev miui version is likely to be the future stable miui version, and the stable version will basically not be updated after the openning unless it is the next Android version 11 etc. 

So could Xiaomi tell us in the issues what changed in the fingerprint drivers K20P and Mi9 Q Sources in the dev (maybe future stable version)

Thanks a lot

shandongtlb, 2019-12-07:
I understand your developers' hard work very much, but I'd like to tell us how to fix or patch it. Thank you

RuyimgByCN, 2019-12-07:
miui11 dev指纹什么时候开放？

Kugumin, 2019-12-07:
Xiaomi, gib working sources for FOD in Dev ROMs plox

LenHackintosh, 2019-12-16:
so didn't get any response?

Kugumin, 2019-12-21:
Seems like Xiaomi doesn't care about that

ivan19871002, 2019-12-23:
after other 8150 device release stable verion.  We will release next kernel

closed 2019-12-23 by ivan19871002

shandongtlb, 2019-12-23:
> after other 8150 device release stable verion. We will release next kernel

thanks XD