MiCode/Xiaomi_Kernel_OpenSource #1181: Redmi Note 7 pro configs

Z3wn, 2019-12-21:
Redmi Note 7 pro arm64 configs files are missing to compile kernel

PIPIPIG233666, 2019-12-21:
Ur _____ is missing to compile kernel

Z3wn, 2019-12-22:
What do you mean? I can't find defconfig to compile violet kernel

shandongtlb, 2019-12-22:
open your eye
arch/arm64/configs/vendor/violet-perf_defconfig

closed 2019-12-23 by ivan19871002

Z3wn, 2019-12-24:
@shandongtlb I got compile error with that defconfig file!!

PIPIPIG233666, 2019-12-24:
Ur problem

Z3wn, 2019-12-24:
@PIPIPIG233666 I got this error.Plz help!!
root@kali:~/violet-p-oss# mkdir out
root@kali:~/violet-p-oss# export ARCH=arm64
root@kali:~/violet-p-oss# export SUBARCH=arm64
root@kali:~/violet-p-oss# export CROSS_COMPILE=/root/violet-p-oss/toolchain/bin/aarch64-linux-android-
root@kali:~/violet-p-oss# make vendor/violet-perf_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
root@kali:~/violet-p-oss# make -j4
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  UPD     include/config/kernel.release
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTCC  scripts/dtc/flattree.o
  SHIPPED scripts/genksyms/parse.tab.c
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/parse.tab.h
  HOSTCC  scripts/genksyms/parse.tab.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/dtc/fstree.o
  CC      kernel/bounds.s
  MKELF   scripts/mod/elfconfig.h
  CHK     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/dtc/data.o
  UPD     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
  CHK     include/generated/timeconst.h
  UPD     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  CC      arch/arm64/kernel/asm-offsets.s
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/mod/modpost.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/selinux/genheaders/genheaders
  CHK     include/generated/asm-offsets.h
  UPD     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
In file included from scripts/selinux/genheaders/genheaders.c:19:
./security/selinux/include/classmap.h:247:2: error: #error New address family defined, please update secclass_map.
  247 | #error New address family defined, please update secclass_map.
      |  ^~~~~
make[3]: *** [scripts/Makefile.host:102: scripts/selinux/genheaders/genheaders] Error 1
make[2]: *** [scripts/Makefile.build:678: scripts/selinux/genheaders] Error 2
make[1]: *** [scripts/Makefile.build:678: scripts/selinux] Error 2
make[1]: *** Waiting for unfinished jobs....
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LDS     arch/arm64/kernel/vdso/vdso.lds
  VDSOA   arch/arm64/kernel/vdso/gettimeofday.o
  VDSOA   arch/arm64/kernel/vdso/note.o
  SHIPPED scripts/dtc/dtc-lexer.lex.c
  SHIPPED scripts/dtc/dtc-parser.tab.h
  SHIPPED scripts/dtc/dtc-parser.tab.c
  VDSOA   arch/arm64/kernel/vdso/sigreturn.o
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/mod/file2alias.o
  VDSOL   arch/arm64/kernel/vdso/vdso.so.dbg
  VDSOSYM include/generated/vdso-offsets.h
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
  HOSTLD  scripts/mod/modpost
make: *** [Makefile:581: scripts] Error 2
root@kali:~/violet-p-oss#

luk1337, 2019-12-24:
>root@kali
kek

shandongtlb, 2019-12-25:
> @shandongtlb I got compile error with that defconfig file!!

This is your misoperation

For msm-4.14 like Mi 9
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone

Z3wn, 2019-12-25:
`For msm-4.14 like Mi 9`

Z3wn, 2019-12-25:
> For msm-4.14 like Mi 9
@shandongtlb I mean note 7 pro(violet)

shandongtlb, 2019-12-25:
> > For msm-4.14 like Mi
> 
> @shandongtlb I mean note 7 pro(violet). Not Mi 9
我真的不知道该说什么好
能不能用你自己试试不就知道了么
我都说了源码是4.14的话就照着米9的教程来
咋这么倔呢

Z3wn, 2019-12-26:
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone

@shandongtlb I can't download

PIPIPIG233666, 2019-12-26:
@Z3wn Sign up...

shandongtlb, 2019-12-26:
自己去想办法
不要总说源码有错误，80%都是自己的问题

Z3wn, 2019-12-29:
[image]
@shandongtlb Any help!!

shandongtlb, 2019-12-29:
root

shandongtlb, 2019-12-29:
And don’t ask me anymore,you have to learn to read the error log yourself