MiCode/Xiaomi_Kernel_OpenSource #1194: When raphael-q-oss?

sar-gib-raphaelqoss, 2020-02-07:
The K20 Pro codename is present twice in the index, one time for the Pie branch and another time for the cepheus-q-oss branch (???), but there is still not a raphael-q-oss branch to this day (even though the Android 10 based MIUI 11 stable update released for the Chinese model many months ago).
Could you please release the sources to the public or at least give information about when it will actually happen?
Thank you

PIPIPIG233666, 2020-02-07:
cepheus one includes raphael changes, can u read?
```
Kernel: Xiaomi kernel changes for Mi 9 and Redmi K20 Pro Android Q

The Patch based on QualComm release TAG: LA.UM.8.1.r1-08700-sm8150.0
The kernel config file used is cepheus_user_defconfig and raphael_user_defconfig

Change-Id: I1eccfd9f987dd38aaf308c41db067be5f0da9183

```

sar-gib-raphaelqoss, 2020-02-07:
I did read that and I in fact wrote about it in the original message, but still the two devices should be separated and each one should have its own tag.