MiCode/Xiaomi_Kernel_OpenSource #1198: olive-p-oss not compile standartalone kernel

AltsecDEV, 2020-02-16:
How to fix this error

make -j$(nproc) O=out CFLAGS=-Wall -Wformat=0 CONFIG_BUILD_ARM64_DT_OVERLAY=y REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang CLANG_TRIPLE=aarch64-linux-gnu- 2>&1 | tee kernel.log
make[1]: вход в каталог «/home/altsec/olive/out»
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  Using .. as source for kernel
  CHK     include/generated/utsrelease.h
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    ../scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  CC      drivers/media/platform/msm/vidc_3x/msm_vidc_common.o
../drivers/media/platform/msm/vidc_3x/msm_vidc_common.c:733:2: error: function definition is not allowed here
        {
        ^
../drivers/media/platform/msm/vidc_3x/msm_vidc_common.c:742:24: error: use of undeclared identifier 'put_inst_helper'; did you mean 'uevent_helper'?
        kref_put(&inst->kref, put_inst_helper);
                              ^~~~~~~~~~~~~~~
                              uevent_helper
../include/linux/kobject.h:37:13: note: 'uevent_helper' declared here
extern char uevent_helper[];
            ^
2 errors generated.

Aquarius223, 2020-03-02:
Please add a string at the end of "olive-perf_defconfig"
# CONFIG_ARM64_ERRATUM_843419 is not set
Try to compile the kernel later, hope it can help you

Aquarius223, 2020-03-02:
Please add the string "CONFIG_ARM64_ERRATUM_843419=n" at the end of "olive-perf_defconfig"
Try to compile the kernel later, hope it can help you

dmitrypershin, 2020-04-06:
olive and olivewood this is garbage collection and not code. xiaomi on top of its arrogance. only ZTE saw such garbage code once

dmitrypershin, 2020-04-06:
> How to fix this error
>
> make -j$(nproc) O=out CFLAGS=-Wall -Wformat=0 CONFIG_BUILD_ARM64_DT_OVERLAY=y REAL_CC=${PWD}/toolchains/llvm-Snapdragon_LLVM_for_Android_6.0/prebuilt/linux-x86_64/bin/clang [...]

you still have to collect wifi and sound (which the code is naturally with errors), and then also look for the cause of the sensors not working .. and these are all visible problems at first glance ..

amir28, 2022-07-17:
I had this problem in compiling Linux kernel for another device. I found out it's because clang does not support nested functions. So can easily fixed it with moving nested functions to file scope(out of parent function). and recommended to change it's name to more specific one.
see https://stackoverflow.com/questions/43842531/compiling-nested-functions-with-clang-versus-gcc 
or can use gcc for compiling kernel