MiCode/Xiaomi_Kernel_OpenSource #1199: umi-q-oss will not build fails on DTC

mrslezak, 2020-02-20:
Using GCC10 and fixed several source bugs.  Fails everytime at the DTC build step.  Something missing in the source?

make[3]: *** No rule to make target 'arch/arm64/boot/dts/qcom/apq8016-sbc.dtb', needed by '__build'.  Stop.

Off others advice I upstreamed DTC completely.  No change. Updated DTC from 1.4.6 (usual Xiaomi requirement) to 1.4.7-1  same error.

Can build Q Mi9 and Mi9T Pro kernels on the same machine with no issues.

https://github.com/mrslezak/umi-q-oss/ is the source directory.  Build script "Build GCC10 Umi.sh" there as well.

mrslezak, 2020-02-20:
Seriously Xiaomi?  The device trees are now not included in the source repo???  Who had this great idea... https://github.com/MiCode/kernel_devicetree/tree/umi-q-oss

LenHackintosh, 2020-02-21:
the source has been dead in 11.0.7 stable miui...........only can boot on 11.0.4 and 11.0.5 stable，so give it up.

ivan19871002, 2020-02-21:
> Seriously Xiaomi? The device trees are now not included in the source repo??? Who had this great idea... https://github.com/MiCode/kernel_devicetree/tree/umi-q-oss - and there's some simple undocumented way to add this to the rest of the source or what? We have to make our own Kconfigs and Makefiles?

Qcom 's idea

closed 2020-02-21 by ivan19871002