MiCode/Xiaomi_Kernel_OpenSource #1200: 7pro不能挂载cifs，内核不支持

zixijian, 2020-02-23:
之前的老手机红米note4x都支持的，方便挂载小米路由器硬盘用。
```
Last login: Sun Feb 23 10:05:07 2020 from 127.0.0.1
Debian GNU/Linux 9 (stretch) [running via Linux Deploy]
cd quark/.t
7pro:~# cd quark/.t
7pro:~/quark/.t# cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   ramfs
nodev   bdev
nodev   proc
nodev   cpuset
nodev   cgroup
nodev   cgroup2
nodev   tmpfs
nodev   configfs
nodev   debugfs
nodev   tracefs
nodev   sockfs
nodev   dax
nodev   bpf
nodev   pipefs
nodev   devpts
        ext3
        ext2
        ext4
        vfat
        msdos
nodev   ecryptfs
nodev   sdcardfs
        fuseblk
nodev   fuse
nodev   fusectl
        f2fs
nodev   pstore
nodev   selinuxfs
        exfat
nodev   functionfs
```

shandongtlb, 2020-02-23:
只要不是开不了机这种问题小米是不会更新的，也没必要更新，自己想办法移植吧

zixijian, 2020-02-23:
> 只要不是开不了机这种问题小米是不会更新的，也没必要更新，自己想办法移植吧

我只是想跟着ota更新，现在都没有一个好的版本，估计到停更也改善不了什么实质性的东西😳

shandongtlb, 2020-02-24:
感觉ota从来都不更新内核，不出现严重bug是不会更的，想用只能自己去移植

zixijian, 2020-02-24:
> 感觉ota从来都不更新内核，不出现严重bug是不会更的，想用只能自己去移植

cpu调度不是内核问题么，从我上个红米note4x到7pro，播放视频时常都是音画不同步，蓝牙看视频更严重，以前的红米note2都没有这种情况。手机内自带反馈石沉大海，反正开发人员都用苹果就是了。

sekaiacg, 2020-02-28:
内核没有开启cifs支持：
# CONFIG_CIFS is not set
你可以自己开启然后编译。

zixijian, 2020-02-29:
> 内核没有开启cifs支持：
> `# CONFIG_CIFS is not set`
> 你可以自己开启然后编译。

感谢指点，回头不行就自己编译了，反正看情况7pro已经成为孤儿机型。

closed 2020-02-29 by zixijian