MiCode/Xiaomi_Kernel_OpenSource #1201: umi-q-oss does not compile, even with Qualcomm repos merged into source

mrslezak, 2020-02-24:
I believe there is source missing to build umi-q-oss.  I found the Makefile in Xiaomi_Kernel_OpenSource arch/arm64/boot/dts/qcom has the latest commit removing Kona and other sm8250 portions of the code.  I moved these from https://github.com/MiCode/kernel_devicetree/tree/umi-q-oss/qcom into the repo but the Makefile there only builds what is in MiCode, so I merged the Makefiles together.  After merging all portions of the Qualcomm source in (techpack audio, data_kernel, qcacld), there is still a build error that cannot be fixed relating to DTC compiling the DTS modules: make[3]: *** No rule to make target 'arch/arm64/boot/dts/qcom/kona.dtb', needed by 'arch/arm64/boot/dts/qcom/cmi-sm8250-overlay.dtbo'.  Stop.
Merged code is all here: https://github.com/mrslezak/umi-q-oss, please advise.   I have compared against sm8150 code that all compiles and cannot isolate any difference that would cause this.

closed 2020-02-27 by mrslezak

nakixii, 2020-08-12:
Have you solved this problem?

mrslezak, 2020-08-12:
Yeah look in my GitHub repo mrslezak umi-q-oss

zixijian, 2021-01-02:
我在红米note9pro gauguin 本机上编译内核遭遇了类似问题。
更换gcc4.8、7.3、8.0均不能成功，不知道内核代码还需要做什么操作。